hifiberry/hifiberry-os #5: airplay

fabvis, 2019-09-27:
airplay don't work with the last version of HiFiBerry OS
airplay is visible, but no sound
(pi3 & amp+)

hifiberry, 2019-09-27:
What is the version you are using. Just login and run

cat /etc/hifiberry.version

Als post the output of

amixer

Do you use it with the Amp+ or Amp2?

fabvis, 2019-09-27:
Amp+

OS version 20190925

# amixer
Simple mixer control 'Master',0
  Capabilities: volume volume-joined
  Playback channels: Mono
  Capture channels: Mono
  Limits: 0 - 248
  Mono: 0 [0%]
Simple mixer control 'DSPVolume',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 230 [90%]
  Front Right: 230 [90%]
Simple mixer control 'Channels',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 248
  Front Left: 218 [88%]
  Front Right: 218 [88%]
Simple mixer control 'Softvol',0
  Capabilities: volume
  Playback channels: Front Left - Front Right
  Capture channels: Front Left - Front Right
  Limits: 0 - 255
  Front Left: 255 [100%]
  Front Right: 255 [100%]

hifiberry, 2019-09-27:
Ok, I need more information ;-)

Now please post the output of the following

cat /etc/hifiberry.state

cat /etc/shairport-sync.conf 

journalctl -u shairport-sync | tail

fabvis, 2019-09-27:
cat /etc/hifiberry.state
CURRENT_MIXER_CONTROL="Master"
CURRENT_CARD="Amp+"
CURRENT_SYSNAME="Bouboule"

cat /etc/shairport-sync.conf
general =
{
  name = "Bouboule";
};

alsa = {
  mixer_control_name = "Master";
};

metadata =
{
  enabled = "yes";
  include_cover_art = "yes";
  pipe_name = "/tmp/shairport-sync-metadata";
  pipe_timeout = 5000;
};

sessioncontrol =
{
  run_this_before_play_begins = "/opt/hifiberry/bin/write-status START AIRPLAY";
  run_this_after_play_ends = "/opt/hifiberry/bin/write-status STOP AIRLAY"; 
  wait_for_completion = "no";
  allow_session_interruption = "yes";
  session_timeout = 20;
};

journalctl -u shairport-sync | tail
-- Logs begin at Thu 2019-02-14 10:12:06 UTC, end at Fri 2019-09-27 11:01:55 UTC. --
Sep 27 10:45:56 Bouboule systemd[1]: Starting Shairport-sync...
Sep 27 10:46:01 Bouboule systemd[1]: Started Shairport-sync.
Sep 27 10:47:31 Bouboule shairport-sync[288]: Execution of on-start command failed to start
Sep 27 10:54:28 Bouboule shairport-sync[288]: Execution of on-stop command failed to start
Sep 27 10:54:34 Bouboule shairport-sync[288]: Execution of on-start command failed to start
Sep 27 10:58:42 Bouboule shairport-sync[288]: Execution of on-stop command failed to start
Sep 27 10:58:57 Bouboule shairport-sync[288]: Execution of on-start command failed to start
Sep 27 10:59:16 Bouboule shairport-sync[288]: Execution of on-stop command failed to start

hifiberry, 2019-09-27:
Ah, now I see it. Master volume is set to 0. You should be able to change volume directly from the iPhone.
Alternatively, you can use "alsamixer" on HiFiBerryOS

fabvis, 2019-09-27:
also no sound if I change volume on iPhone ...

where to change alsamixer ? here ?

alsa = {
mixer_control_name = "Master";
};

hifiberry, 2019-09-27:
No, do not edit the file, just type alsamixer

fabvis, 2019-09-27:
yeah ! 🤘🏼

thank you !

hifiberry, 2019-09-27:
It's strange that volume control isn't working from the iPhone. Next version will also allow to control the master volume from a web interface...

fabvis, 2019-09-27:
if I reboot, master volume is 100%
and when I play a file via airplay, master volume come to 0%

fabvis, 2019-09-27:
and when I change the volume via iTunes, back to 0%

fabvis, 2019-09-27:
and when I connect airplay from another device, back to 0%

fabvis, 2019-09-27:
volume controle work fine with DAC+

hifiberry, 2019-09-30:
Ah, I remember that Shairport is picky with the volume control. Can you try editing /etc/shairport-sync.conf and change the volume control to 

alsa = {
mixer_control_name = "Softvol";
};

fabvis, 2019-09-30:
hi !
for info, if I made /bin/shairport-sync -vvv
I've found this:

Mixer device name is "default".
Mixer Control name is "Master".
note: the hardware mixer specified -- "Master" -- does not have a dB volume scale.

I try now with softvol

fabvis, 2019-09-30:
with softvol, volume control is ok from iTunes !
master is at 100%, should I lower it a little ?

hifiberry, 2019-09-30:
Hi,

you can lower it a little so that the maximum volume is ok for you. We'll change volume control to softvol in the next HiFiBerryOS release.

Best regards,
Daniel

fabvis, 2019-09-30:
thank you for your help Daniel
best regards

closed 2019-09-30 by hifiberry