hifiberry/hifiberry-os #6: Bluetooth not working

Prog72, 2019-10-14:
Logitech Media Server works perfectly but I'm unable to get Bluetooth audio working at all. I can connect my phone (Samsung Galaxy S8) to HiFiBerry but get no audio, just the following error over and over again.

     Bluetooth: hc10: Frame reassembly failed (-84)

I've tried with both a Pi 3B and a Pi 3B+ and both yield the same result.

Output of systemctl status bluetoothd with the phone connected and attempting to play audio over Bluetooth is as follows:

     ● bluetoothd.service - Bluetooth demon

     Loaded: loaded (/usr/lib/systemd/system/bluetoothd.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2019-10-14 14:19:45 UTC; 4min 18s ago
     Main PID: 174 (bluetoothd)
     Tasks: 1 (limit: 2143)
     Memory: 2.8M
     CGroup: /system.slice/bluetoothd.service
     └─174 /usr/libexec/bluetooth/bluetoothd -f /etc/bluetooth/main.conf -P sap,hostname

     Oct 14 14:19:58 Lounge bluetoothd[174]: Bluetooth management interface 1.14 initialized
     Oct 14 14:20:00 Lounge bluetoothd[174]: Player registered: sender=:1.12 path=/_1_9
     Oct 14 14:20:01 Lounge bluetoothd[174]: Endpoint registered: sender=:1.14 p    ath=/hci0/A2DP/MPEG24/Sink/1
     Oct 14 14:20:01 Lounge bluetoothd[174]: Endpoint registered: sender=:1.14 path=/hci0/A2DP/SBC/Sink/1
     Oct 14 14:20:03 Lounge bluetoothd[174]: Player registered: sender=:1.12 path=/_1_17
     Oct 14 14:20:03 Lounge bluetoothd[174]: Player registered: sender=:1.12 path=/_1_18
     Oct 14 14:20:14 Lounge bluetoothd[174]: Player registered: sender=:1.12 path=/_1_20
     Oct 14 14:23:49 Lounge bluetoothd[174]: Endpoint registered: sender=:1.14 path=/hci0/A2DP/MPEG24/Sink/2
     Oct 14 14:23:50 Lounge bluetoothd[174]: /org/bluez/hci0/dev_C8_D7_B0_09_2C_F6/fd0: fd(21) ready
     Oct 14 14:23:53 Lounge bluetoothd[174]: Control: Refusing unexpected connect

hifiberry, 2019-10-15:
Hi,

the 3B (without the +) might be problematic. Can you try the following on the Pi3B+:
 Replace the last line in /opt/btspeaker by the following:

/usr/bin/hciattach -n /dev/ttyAMA0 bcm43xx 3000000 flow - $BDADDR

Then reboot and try again.

This won't work on the Pi3B.

hifiberry, 2019-10-21:
Any feedback on this?

hifiberry, 2019-10-30:
Please try the latest HiFiBerryOS build. You can upgrade by logging in and running
/opt/hifiberry/bin/update

hifiberry, 2019-10-30:
PS: For BT audio, you need to use the Pi3B+ (or 0W or Pi 4) as the Pi3's hardware design makes it almost impossible to have stable high-bandwidth BT connections.

closed 2019-11-21 by hifiberry