hifiberry/hifiberry-os #7: RP3b+AMP2 - Squeezelite output distorted after a while

tacshen, 2019-10-16:
Using a RP3b+ and AMP2 with Squeezelite.

Sound distortion occurs after a while (couple hours) during LMS playback or a long pause. It sounds like the sample rate is too low for the played media. Always occurs, always fixable by turning the device off in LMS and on again.

Sound from other sources (e.g. local play) is not affected, only LMS playback.

People with a similar problem:
https://www.max2play.com/en/forums/topic/squeezelite-always-needs-to-be-restarted-after-non-use/
https://www.max2play.com/en/forums/topic/distorted-sound-on-rpi3/
https://github.com/sixeyed/squeezelite/issues/31

# systemctl status squeezelite
● squeezelite.service - Squeezelite
   Loaded: loaded (/usr/lib/systemd/system/squeezelite.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-10-16 09:50:41 UTC; 6h ago
  Process: 345 ExecStartPre=/usr/sbin/alsactl --file /tmp/sqvol store (code=exited, status=0/SUCCESS)
  Process: 346 ExecStartPre=/bin/bash -c /usr/bin/systemctl set-environment MIXER=`/opt/hifiberry/bin/volume-control-name` (code=exited, status=0/SUCCESS)
  Process: 350 ExecStartPost=/usr/bin/sleep 5 (code=exited, status=0/SUCCESS)
  Process: 355 ExecStartPost=/usr/sbin/alsactl --file /tmp/sqvol restore (code=exited, status=0/SUCCESS)
 Main PID: 349 (squeezelite)
    Tasks: 4 (limit: 2143)
   Memory: 16.6M
   CGroup: /system.slice/squeezelite.service
           └─349 /usr/bin/squeezelite -o default -M HiFiBerry -N /var/squeezelite/squeezelite.name -V Digital

Oct 16 09:50:36 HiFiBerry systemd[1]: Starting Squeezelite...
Oct 16 09:50:41 HiFiBerry alsactl[355]: /usr/sbin/alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock PLL:0' : Device or resource busy
Oct 16 09:50:41 HiFiBerry alsactl[355]: /usr/sbin/alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock DSP:0' : Device or resource busy
Oct 16 09:50:41 HiFiBerry alsactl[355]: /usr/sbin/alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock DAC:0' : Device or resource busy
Oct 16 09:50:41 HiFiBerry systemd[1]: Started Squeezelite.
Oct 16 09:50:59 HiFiBerry squeezelite[349]: [09:50:59.606786] _write_frames:615 mmap_commit error

Viti0, 2019-10-21:
I circumvented the issue by setting squeezelites output device explicitly instead of default.

```
vi /usr/lib/systemd/system/squeezelite.service
ExecStart=/usr/bin/squeezelite -o sysdefault:CARD=sndrpihifiberry -M HiFiBerry -N /var/squeezelite/squeezelite.name -V ${MIXER}
systemctl daemon-reload
systemctl restart squeezelite.service
```

No further sound distortion occurred after this. This had an effect on how volume is managed however, as the initiation of mixer failed and a reversion to software volume is in place (Cancelling the -V ${MIXER} option)

```
# systemctl status squeezelite
● squeezelite.service - Squeezelite
   Loaded: loaded (/usr/lib/systemd/system/squeezelite.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-10-20 10:18:41 UTC; 20h ago
  Process: 2770 ExecStartPre=/usr/sbin/alsactl --file /tmp/sqvol store (code=exited, status=0/SUCCESS)
  Process: 2771 ExecStartPre=/bin/bash -c /usr/bin/systemctl set-environment MIXER=`/opt/hifiberry/bin/volume-control-name` (code=exited, status=0/SUCCESS)
  Process: 2775 ExecStartPost=/usr/bin/sleep 5 (code=exited, status=0/SUCCESS)
  Process: 2780 ExecStartPost=/usr/sbin/alsactl --file /tmp/sqvol restore (code=exited, status=0/SUCCESS)
 Main PID: 2774 (squeezelite)
    Tasks: 4 (limit: 2143)
   Memory: 16.7M
   CGroup: /system.slice/squeezelite.service
           └─2774 /usr/bin/squeezelite -o sysdefault:CARD=sndrpihifiberry -M HiFiBerry -N /var/squeezelite/squeezelite.name -V Digital

Oct 20 10:18:36 HiFiBerry systemd[1]: Starting Squeezelite...
Oct 20 10:18:36 HiFiBerry squeezelite[2774]: [10:18:36.674686] mixer_init_alsa:869 attach error: Invalid argument
Oct 20 10:18:36 HiFiBerry squeezelite[2774]: [10:18:36.675031] output_init_alsa:989 Initialization of mixer failed, reverting to software volume
Oct 20 10:18:41 HiFiBerry alsactl[2780]: /usr/sbin/alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock PLL:0' : Device or resource busy
Oct 20 10:18:41 HiFiBerry alsactl[2780]: /usr/sbin/alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock DSP:0' : Device or resource busy
Oct 20 10:18:41 HiFiBerry alsactl[2780]: /usr/sbin/alsactl: set_control:1461: Cannot write control '2:0:0:Max Overclock DAC:0' : Device or resource busy
Oct 20 10:18:41 HiFiBerry systemd[1]: Started Squeezelite.
```

Now the system wide volume is not controlled by squeezelite anymore. I have to use alsamixer to set Digital to 100% and then use squeezelites volume bar in my application. As LMS is the only system I'm using, I'm quite happy as it is now, you just need to be aware of it.

hifiberry, 2019-10-21:
Hi,

doing this will block all other players.  The service might not even start after a reboot.

Best regards,
Daniel

hifiberry, 2019-10-25:
Hmm, seems that on other installations that might have been cause by mmap().
Can you try the following command line:

/usr/bin/squeezelite -o default -M HiFiBerry -N /var/squeezelite/squeezelite.name -V Digital -a 2048:3446:24_3:0

Viti0, 2019-10-26:
Tried the line, distortion still occurred. Additionally, files with 24/192 could not be played at normal speed anymore :-(

hifiberry, 2019-10-30:
Ok, I think the problem was the resampler. This configuration has been changed now.

Please try the latest HiFiBerryOS build. You can upgrade by logging in and running
/opt/hifiberry/bin/update

hifiberry, 2019-10-30:
BTW: If you made any changes to the system, I recommend to create a backup before upgrading.

Viti0, 2019-10-31:
Although the complete reset of all configuration settings during the update process (hostname, ssh fingerprint, wifi, LMS speaker name, volume, turned off/on services) is a pain and far from an ideal user experience, it seems that the issue is resolved with the changes you made.

hifiberry, 2019-10-31:
This is strange, as settings should have been taken over. Do you know what release you have used before?

hifiberry, 2019-10-31:
BTW: You can (and should) change all settings now using the new Web UI (just connect your browser to the system)

Viti0, 2019-10-31:
Yes, I still have a second device up and running on the previous version.

```
# cat /etc/os-release 
NAME=Buildroot
VERSION=2019.08-rc3
ID=buildroot
VERSION_ID=2019.08-rc3
PRETTY_NAME="Buildroot 2019.08-rc3"
```

I'd like to change all settings using the Web UI, but cannot seem to find it. Pointing to the IP at port 80, I only see the following. Port 9000 is the usual ympd, which port are you referring to? Which service should be up and running?

[image]

I've used the self-removing config file for now, but maybe I'll just wipe the system and do a clean install.

hifiberry, 2019-10-31:
Hi,

the last version had only this display. With the new release you can change the settings in the UI.
What's the output of
 cat /etc/hifiberry.version of this system?

hifiberry, 2019-10-31:
Please also post the content of 

/opt/hifiberry/etc/config-files

Of the old system

Viti0, 2019-10-31:
Content of the new system:

```
# cat /etc/hifiberry.version
20191030
```

Content of the old system
```
# cat /opt/hifiberry/etc/config-files
/etc/hifiberry/initial-volume
/etc/systemname
/etc/bluetooth/main.conf
/etc/shairport-sync.conf
/etc/uuid
/etc/mpd.conf
/var/squeezelite/squeezelite.name
/etc/audiocontrol2.conf
/etc/spotifyd.conf
/etc/beocreate
```

hifiberry, 2019-10-31:
Ok. I see why the settings haven't been taken over correctly. Unfortunately this is something we can't fix anymore. While it would be possible, to change some files to take over the settings, the easiest way will be to upgrade the system and do the configuration changes (hostname, services) in the UI. This will take only a minute or so as you don't have to edit files anymore.
The current update was a major upgrade that changed a lot on the system. I hope future upgrades will run a lot smoother and ideally won't need any re-configuration again. It will become better and hopefully with the next version you should even be able to do upgrades from the UI.

closed 2019-10-31 by hifiberry