hifiberry/hifiberry-os #8: RP3b+AMP2 - Digital volume spikes during playback for a fraction of a second

Viti0, 2019-10-21:
I have two RP3b+AMP2 with one speaker each (R, L) with LMS in sync. During LMS playback, audio volume spikes occur, which are quite annoying. Sometimes not for hours, sometimes multiple times a minute. In alsamixer, Digital volume will increase from e.g. 20 to 36 and back within one second. My assumption is that another application is attempting to set the volume but is overwritten by squeezelite again, but I could not figure out which one. I turned off all the other applications in hifiberry.conf, it had no effect.

```
# cat /etc/hifiberry.conf
system_name="HiFiBerry"

wifi_ssid=NETWORK
wifi_psk=NETPASS
wifi_country=DE
wifi_enable=1

spotify_enable=0
spotify_user=user
spotify_password=pass

airplay_enable=0

roon_enable=0

squeezebox_enable=1

bluetooth_enable=0

sound_initialvolume="30%"

auto_pause=0

scrobbler_network=librefm
scrobbler_apikey=hifiberryos
scrobbler_apisecret=hifiberryiscool
scrobbler_username=myuser
scrobbler_password=mypassword
```

The problem was circumvented by letting squeezelite use software volume rather than giving it access to Digital. Digital is now set to 100% at all times and no further spikes occurred. For my use case (as I use LMS exclusively) it's fine this way.

hifiberry, 2019-10-21:
This is a problem with a bug in Spotifyd. Changing the volume control "by hand" might break other parts of the system, therefore I would not recommend this.  Try disabling spotify

systemctl disable spotify

Viti0, 2019-10-22:
Ok, due to your recommendation I reverted my changes and disabled spotify instead, so far I have not encountered any volume spikes, so as you said spotify must have been the culprit. This however brought back the other issue I had with the distortions after a long pause (https://github.com/hifiberry/hifiberry-os/issues/7).

hifiberry, 2019-10-23:
We'll see if we can find a solution for the Squeezelite stuff. I will close this.

closed 2019-10-23 by hifiberry