hifiberry/hifiberry-os #9: Bluetooth not working with pi zero w

cnieves1, 2019-10-22:
Hi, 
I dowloaded and installed your pi zero w image in a pi zero w v1.1.
I managed to get lms working correctly, but I look for Bluetooth devices, I can't see Hifiberry.
```
# systemctl status bluetoothd
● bluetoothd.service - Bluetooth demon
   Loaded: loaded (/usr/lib/systemd/system/bluetoothd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-10-22 21:23:59 UTC; 8min ago
 Main PID: 132 (bluetoothd)
   Memory: 2.3M
   CGroup: /system.slice/bluetoothd.service
           └─132 /usr/libexec/bluetooth/bluetoothd -f /etc/bluetooth/main.conf -P sap,hostname

Oct 22 21:24:02 HiFiBerry bluetoothd[132]: Bluetooth daemon 5.50
Oct 22 21:24:03 HiFiBerry bluetoothd[132]: Starting SDP server
Oct 22 21:24:03 HiFiBerry bluetoothd[132]: Excluding (cli) hostname
Oct 22 21:24:03 HiFiBerry bluetoothd[132]: Bluetooth management interface 1.14 initialized
Oct 22 21:24:03 HiFiBerry bluetoothd[132]: Failed to set privacy: Rejected (0x0b)
Oct 22 21:23:59 HiFiBerry systemd[1]: Started Bluetooth demon.
Oct 22 21:24:29 HiFiBerry bluetoothd[132]: Endpoint registered: sender=:1.11 path=/hci0/A2DP/MPEG24/Sink/1
Oct 22 21:24:29 HiFiBerry bluetoothd[132]: Endpoint registered: sender=:1.11 path=/hci0/A2DP/SBC/Sink/1
Oct 22 21:24:33 HiFiBerry bluetoothd[132]: Player registered: sender=:1.9 path=/_1_13
Oct 22 21:24:54 HiFiBerry bluetoothd[132]: Player registered: sender=:1.9 path=/_1_18

```
and:

```
# hcitool dev
Devices:
	hci0	B8:27:EB:22:17:BA
# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:22:17:BA  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:766 acl:0 sco:0 events:49 errors:0
	TX bytes:1536 acl:0 sco:0 commands:49 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'HiFiBerry'
	Class: 0x040414
	Service Classes: Rendering
	Device Class: Audio/Video, Loudspeaker
	HCI Version: 4.1 (0x7)  Revision: 0x168
	LMP Version: 4.1 (0x7)  Subversion: 0x2209
	Manufacturer: Broadcom Corporation (15)

```

Any hints? 
Thanks

hifiberry, 2019-10-23:
Looks fine. However, there is a bug in the Bluetooth implementation right now. I recommend to check the new release that should be available in the next 1-2 weeks.

hifiberry, 2019-10-30:
Please try the latest HiFiBerryOS build. You can upgrade by logging in and running
/opt/hifiberry/bin/update

cnieves1, 2019-10-30:
Mmm... not so simple. I run the update, but after rebooting, it does never connect again to my wifi.
I think the SD card had only one 1.1GB partition with /etc and so on, before the update. After the update, it seems it has two 1.1GB partitions, both with /etc and so on... I copied /etc/hifiberry.conf from one to the other (where it does not exist), but it does not connect yet...
I have no more time today to look into this, but I will come back...

hifiberry, 2019-10-31:
/etc/hifiberry.conf is not used during normal operations anymore. You can configure WiFi with the WiFi setup agent now:
https://www.hifiberry.com/docs/software/hifiberryos-configure-wifi/

cnieves1, 2019-10-31:
IIRC, there was no wifi named HifiBerry... I think I will try to flash a SD image with the new version...

hifiberry, 2019-10-31:
The name should be hifiberry-setup-xxxx. On the Pi Zero, it can take quite a long time until the system is up after an initial installation - it can take up to 10 minutes as the Zero is really slow even when compared to the 3B.
If the system hangs on initial boot (no activity shown on the activity LED), unplug/plug power again. Initial boot seems to hang sometimes, depending on the OS/SD card writer software.

cnieves1, 2019-10-31:
Downloaded the image and flashed. It hanged on initial boot, but at the second or third reboot, the hifiberry-setup wifi showed and I could configure the Wifi.
I can now see it when looking for bluetooth devices. I tried to connect to it, and I got succesfull at the fourth time. I didn't see any error, it just didn't get connected the first times.
Then, I tried to play a local file. I could hear the sound, but there were some pauses every two or three seconds... Could this be improved?
I also noticed that Squeezelite was disabled. Why? Will it be enabled in the next versions?
Thanks!

hifiberry, 2019-11-01:
Squeezelite won't be enabled on the Pi Zero anymore as the Pi Zero just isn't powerful enough for this. We haven't seen the drops here. What exactly do you do to play a local file?

cnieves1, 2019-11-03:
I meant connecting a mobile phone with Hifiberry through Bluetooth and playing an audio file on the phone. I tried again, using the same phone and another different phone. It worked more or less. I could still hear some drops, from both phones, but they were not so frequent, maybe one in 5 to 10 minutes, so it could be something outside Hifiberry...
Please consider this thread closed, as the main issue was not being able to connect through Bluetooth, and this was already solved. Thanks!

hifiberry, 2019-11-03:
Let's still try to figure out what's happening. How far is your player away from your mobile phone?  It might be a reception issue or some application that's doing something in the background on the Pi. As the Pi Zero has only a single core CPU, I guess

cnieves1, 2019-11-03:
Only 1-2 meters away. The first time I tried was in a environment with a lot of wifis. Don't know if It has something to do. This last time with two phones there were no wifis. What can I do to debut this?

hifiberry, 2019-11-03:
Lots of WiFi traffic can have an impact as it's using the same frequency range. We're planning a small bugfix release next week. I'll include some tools to record performance data on the system with this release.

cnieves1, 2019-11-03:
Ok. I will test It!
Regarding performance, I tested, only a bit, squeezelite with the release before this one and it seemed to work fine. Did you experienced drops with It? Are you planning to support any other system with ZeroW? Snapcast for example? Thanks!

hifiberry, 2019-11-23:
Can you try the latest release? You can upgrade using
/opt/hifiberry/bin/update

We don't plan more sources on the Pi Zero. It's just very hard to maintain this as it uses a different CPU architecture and isn't very powerful. So testing needs a lot of time and there are very few Pi Zero users.
I recommend to go for the Pi 3A+ instead if you want to run the "full" version.
However, feel free to experiment by yourself. You can check out the source and build you own customized version. Check out the "developer documentation" here:
https://github.com/hifiberry/hifiberry-os/tree/master/doc

cnieves1, 2019-11-23:
Back on "lots of wifis" environment. Seems I'm having frequent drops. I run "top" on hifiberry, and it shows a cpu load about 22% and about 150MB free memory, so it is not loaded...
Is there any way to increase the audio sound "cache"? 
what about the tools to measure performance data?

cnieves1, 2019-11-23:
the "top" was run when I was playing a song on my mobile while the bluetooth connection was active, so I can hear that song through hifiberry.

hifiberry, 2019-11-23:
Try

 perf top

Do you have a Pi3B+ or Pi4 available? It would be important to see if the same issue also happens with one of these.

cnieves1, 2019-11-23:
# perf top
Error: cycles: PMU Hardware doesn't support sampling/overflow-interrupts. Try 'perf stat'
# 
# perf stat
 Performance counter stats for 'system wide':

         147787.77 msec cpu-clock                 #    1.000 CPUs utilized          
            648177      context-switches          # 4385.886 M/sec                  
                 0      cpu-migrations            #    0.000 K/sec                  
              5878      page-faults               #   39.773 M/sec                  
       47082150267      cycles                    # 318581.135 GHz                    (75.00%)
       11072944358      instructions              #    0.24  insn per cycle           (75.00%)
        1486790178      branches                  # 10060358.340 M/sec                (75.00%)
         224344565      branch-misses             #   15.09% of all branches          (50.00%)

     147.786712893 seconds time elapsed

[No drops this time]

hifiberry, 2019-11-23:
Hmm, the old Pi0 CPU is missing a lot of things :(
The perf stat output doesn't really help. 
I think the best you can do is try a Pi 3A+/B+ or Pi4 and see if you see the same problems.

cnieves1, 2019-11-23:
Performance counter stats for 'system wide':

         126895.18 msec cpu-clock                 #    1.000 CPUs utilized          
            438205      context-switches          # 3453.288 M/sec                  
                 0      cpu-migrations            #    0.000 K/sec                  
              6766      page-faults               #   53.320 M/sec                  
       33226127728      cycles                    # 261839.534 GHz                    (74.99%)
        7606756055      instructions              #    0.23  insn per cycle           (75.00%)
        1028412189      branches                  # 8104434.288 M/sec                 (75.00%)
         158302354      branch-misses             #   15.39% of all branches          (49.99%)

     126.894523994 seconds time elapsed

[drops this time]

Sorry, I don't have any other Pi available...

hifiberry, 2019-11-23:
Unfortunately this output doesn't help as it doesn't show a per-function reporting that the "perf top" command shows.

Unfortunately I can't do anything here :(

cnieves1, 2019-11-23:
I guessed that... Anyway, thanks for your help!

closed 2024-01-12 by hifiberry