hifiberry/hifiberry-os #10: Airplay increases volume to maximum

hifiberry, 2019-10-27:
When using Airplay, it can happen that the volume control increases automatically.
It is not yet clear, when this happens.

hifiberry, 2019-10-29:
Seems to be a bug in the Beocreate web interface.

hifiberry, 2019-10-29:
This has been fixed with a new Beocreate release

closed 2019-10-29 by hifiberry

danbee, 2020-02-06:
I know this is closed, but I'm experiencing this now with the latest version of HifiBerryOS on a Raspberry Pi 3. Sometimes it jumps momentarily and back down again. Sometimes it jumps and doesn't reset until I tweak the volume slider.

hifiberry, 2020-02-06:
You problem has probably nothing to do with this. Try disabling Spotify and check if this still happens. If you have configured a username/password for Spotify, remove this.

danbee, 2020-02-06:
Thank you for your reply. I found the other thread and disabling Spotify does indeed seem to solve this!