hifiberry/hifiberry-os #14: Shutdown and restart not working completely.

x838nwy, 2019-10-31:
Using version 20191030 on a pi4 and Digi+ Pro through wifi

Started up and the screen shows a HifiBerryOS logo and "web interface ready at 192.168.1.14" which is nice.

I can access the interface on Chrome and all seems fine. But when I try to shut down or restart, Chrome displays a graphic and says "Restarting product" but the display on the TV is stuck on the HifiBerryOS logo and "web interface ready at 192.168.1.14" as before. I waited a while (3-4 minutes) and still no change and eventually Chrome displays "Product is unreachable". All the while on the screen, the display is stuck on the "web interface ready..." and I can no longer access the web interface until I unplug/plug the power jack.

The same thing happens without the hdmi cable connected.

*note - I'm not sure if shutdown works when hdmi is not connected. The red light remains on, however.

Thanks

C

tuomashamalainen, 2019-10-31:
We've seen this issue ourselves and are investigating. Thanks.

Tuomas / Bang & Olufsen

lucasso72, 2019-11-10:
Same here, the shutdown definitely doesnt work as it should, it (gracefully) kills the httpd daemon for sure and maybe some other services (I didn't checked before/after number of processes running honestly) but the OS is definitely up&running, if you ping the IP (I'm using RPI 3A+ thus wifi only) it's still alive an replying to ping.
HTH

hifiberry, 2019-11-22:
This should be fixed now. Please check the latest version. You can update from command line using
/opt/hifiberry/bin/update

lucasso72, 2019-11-22:
I confirm it's fixed.

closed 2019-11-22 by hifiberry