hifiberry/hifiberry-os #17: 20191121 Roon client flashes on/off

x838nwy, 2019-11-22:
Downloaded a new version and things are not looking good.
Roon from iOS and also my laptop failed to actually locate the HifiBerry. It is very odd - it appears on the list for perhaps half a second and then just disappears. This flashing continues for quite some time then it just stops.

Happens both over wifi and wired.

Please see
[https://www.dropbox.com/s/z8kyxx9z7j500xx/VID_20191122_205624.mp4?dl=0](url)

hifiberry, 2019-11-22:
Have you upgraded to Roon 1.7? It seems there is a problem with the new Roon version.

hifiberry, 2019-11-22:
What Pi version do you use?

x838nwy, 2019-11-22:
Yes, Roon is 1.7 and I'm using a pi4. Sorry for omitting the information earlier.

hifiberry, 2019-11-22:
Looks like 1.7 triggers a bug in our software. Need to check how this can be fixed.

lucasso72, 2019-11-22:
mmh I'm now on HiFiBerry 20191121 with Roon 1.7, using DAC Pro+ on a Pi 3A+, but I do NOT have the issue.

hifiberry, 2019-11-22:
Seems to happen specifically if the client isn't already registered in Roon.

hifiberry, 2019-11-22:
I pushed a (minimally tested) emergency update to our web site. Try updating via the updater in the UI (System/Software updater) and see if this fixes the issue.

x838nwy, 2019-11-23:
Tired the update through the ui - no luck. Progress completer but the restart did not complete.

Downloaded the image and tried a reinstall. UI worked and I got airplay working. Sadly enabling roon killed everything - airplay stopped working. I have not tested any thing else.

Will investigate further this afternoon my time.

hifiberry, 2019-11-23:
Hmm, as you system seems to behave a bit strange in general, try another SD card. We have seen all kinds of crazy behaviour with corrupted SD cards.

x838nwy, 2019-11-23:
Tried another SD card this one is a Sandisk so it should be reliable (plus it came out of its package like 4 days ago). Starting Roon did not kill it, but restarting from the UI seems to change sources, meaning I need to re-enable.

Please see [https://www.dropbox.com/s/4pit8j3cqandn5i/VID_20191123_150249.mp4?dl=0](url)

I actually have another pi4 & DIGI+ PRO coming in the mail in a couple of days, would be really great if they both work.

hifiberry, 2019-11-23:
The status display of the sources in the UI is partially incorrect after a reboot. However, if you enabled a service, it is still enabled, even if the UI shows it differently.

x838nwy, 2019-11-23:
> The status display of the sources in the UI is partially incorrect after a reboot. However, if you enabled a service, it is still enabled, **even if the UI shows it differently**.

Yeah, I was going crazy over this. Thank you for pointing it out.

hifiberry, 2019-11-23:
Is the Roon client working now?

x838nwy, 2019-11-23:
> Is the Roon client working now?

Sadly I cannot check it right now. Will update as soon as I can.

x838nwy, 2019-11-23:
Wired Roon now appears to work with no instability that I managed to find in normal use (of about 2 hours last night). Wireless it seems flaky, but it might be more a wifi issue particular to my situation.

hifiberry, 2019-11-24:
Yes, WiFi can be a bit problematic if reception isn't perfect.

closed 2019-11-24 by hifiberry