hifiberry/hifiberry-os #18: 20191121 Need to re-enable source at each re-start

x838nwy, 2019-11-22:
I can enable any number of sources but if the unit is restarted, the only source that remains enabled is Bluetooth. Everything else needs to be re-enabled.

hifiberry, 2019-11-22:
What Pi version do you use?

x838nwy, 2019-11-22:
Pi4. My mistake for not letting you know.

hifiberry, 2019-11-22:
Hmm, strange. Seems to work in my installation.  Maybe it's a specific configuration that triggers this. What sound card are you using?

x838nwy, 2019-11-22:
Digi+ on the pi4. I'll try some other configurations and see if this persists.

x838nwy, 2019-11-23:
Okay, I went back to 20191121 and updated from the UI with a screen connected and got found that while it threw an epi after downloading, unpacking and such,  unplugging and re-plugging the power brought it back to life.

An odd thing is that after doing the above, there was no wireless option in the UI after the machine booted. So anyway I set sources to Roon and Airplay(1).  Had not yet tested Roon's functionality but I decided to restart through the UI and the only source that was saved was Bluetooth.

hifiberry, 2019-11-23:
Ok, I think I could reproduce it here. Under some circumstances the sources are shown incorrectly in the UI. In fact the sources are active, but the UI shows them as inactive.

hifiberry, 2019-11-23:
As it is just a cosmetic thing, we'll fix this in the next release.

hifiberry, 2019-11-24:
As we had to do an emergency bugfix release due to another issue, we already included the bugfix for this. Upgrading to the new 20191124 version should fix this.

x838nwy, 2019-11-26:
> As we had to do an emergency bugfix release due to another issue, we already included the bugfix for this. Upgrading to the new 20191124 version should fix this.

Downloaded the image and will give it a go this evening. Sorry been a bit occupied these last few days.

x838nwy, 2019-11-26:
All seems to be working well right now :)

hifiberry, 2019-11-26:
Thanks for the feedback

closed 2019-11-26 by hifiberry