hifiberry/hifiberry-os #23: Bluetooth not connecting to TV - fine on phone

Viti0, 2019-11-26:
RPi 3 B+
AMP2

```
# cat /etc/os-release
NAME=Buildroot
VERSION=2019.08.2
ID=buildroot
VERSION_ID=2019.08.2
PRETTY_NAME="Buildroot 2019.08.2"
```

```
# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2019-11-26 08:12:54 UTC; 10min ago
     Docs: man:bluetoothd(8)
 Main PID: 1252 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 2143)
   Memory: 872.0K
   CGroup: /system.slice/bluetooth.service
           └─1252 /usr/libexec/bluetooth/bluetoothd

Nov 26 08:12:54 HiFiBerry bluetoothd[1252]: Bluetooth daemon 5.50
Nov 26 08:12:54 HiFiBerry systemd[1]: Started Bluetooth service.
Nov 26 08:12:54 HiFiBerry bluetoothd[1252]: Starting SDP server
Nov 26 08:12:54 HiFiBerry bluetoothd[1252]: Bluetooth management interface 1.14 initialized
Nov 26 08:12:54 HiFiBerry bluetoothd[1252]: Player registered: sender=:1.42 path=/_1_25
Nov 26 08:12:54 HiFiBerry bluetoothd[1252]: Player registered: sender=:1.42 path=/_1_16
Nov 26 08:12:54 HiFiBerry bluetoothd[1252]: Player registered: sender=:1.42 path=/_1_19
Nov 26 08:12:54 HiFiBerry bluetoothd[1252]: Player registered: sender=:1.42 path=/_1_18
Nov 26 08:12:59 HiFiBerry bluetoothd[1252]: Endpoint registered: sender=:1.64 path=/hci0/A2DP/MPEG24/Sink/1
Nov 26 08:12:59 HiFiBerry bluetoothd[1252]: Endpoint registered: sender=:1.64 path=/hci0/A2DP/SBC/Sink/1
```

I want to set the HiFiBerry as the sound output device of my TV, but it doesn't pair and connect. Works fine on my Android device though. The TV is a Samsung Q70R. What can I do to resolve this?
[image]
[image]
[image]
[image]

hifiberry, 2019-11-26:
Make sure your Android phone is disconnected. Only one device is allowed to connect at the same time

Viti0, 2019-11-26:
Yes, I disconnected and unpaired the Android device before I tried connecting the phone. This is sadly not the issue.

hifiberry, 2019-11-26:
Unfortunately there is nothing else I can do about this. We don't have this TV here and are not able to debug this.

Viti0, 2019-11-26:
OK. Can I use the TVs TOSLINK somehow to get audio to HiFiBerryOS?

closed 2019-11-26 by Viti0

hifiberry, 2019-11-26:
You can with the Beocreate 4 channel amplifier, it has a TOSLink input. The Amp2 doesn’t have TOSLink.