hifiberry/hifiberry-os #25: Web interface not working (new SD)

x838nwy, 2019-12-19:
I have two pi4 with DiGi+, updated both today and one went through absolutely fine, but the other didn't take. The one that fail was visible in Roon, but streaming would only happen for about 10 seconds and it'd just die and disappear from Roon. It will only reappear after a re-start but still could not stream for more than 10 seconds or so.

For the one that failed, I downloaded the image from the website and proceeded a 'clean' install.

This sadly didn't work. Tried with 2 different cards and unzipped using the 'normal' windows tool and 7-zip. No difference. I can actually stream to it in Roon fine and it does not die after a while but the GUI does not load at all. I get a grey screen with "next step" on the bottom left-hand corner which I cannot click as it is greyed out.

[image]

Please help.

tuomashamalainen, 2019-12-19:
Hi,

sorry about your issue. Can you confirm that this hangup is in the first screen in the setup assistant, i.e. you haven't seen a "welcome" screen yet at this point?

Best,
Tuomas from Bang & Olufsen

x838nwy, 2019-12-19:
Well, very briefly - may be less than half a second -  before getting to this screen, the top menu bar of the gui flahes up and disappears. Other than that, this is all I get, sadly. Certainly no "welcome" screen. Hope this helps.

tuomashamalainen, 2019-12-20:
Thanks. The menu bar disappearing is expected behaviour, it will be hidden when the setup assistant – which is obviously very empty right now – is on screen.

If I may ask one more thing from you to help investigate this:
Please open Developer Tools (three dots in top right -> More Tools -> Developer Tools), go to Console and refresh the page. There will be a bunch of messages there. What I'm specifically interested in are:
- errors (in _red_, other than bluetooth.svg or favicon missing)
- a message that has _header: "setupStatus"_ – what's in its _content_?

Best,
Tuomas

x838nwy, 2019-12-20:
No problem - will do later today.

x838nwy, 2019-12-20:
I have no idead what I'm looking at so here goes:
The stuff in **bold** text is shown in red.

VM225:1 **Uncaught SyntaxError: Unexpected token o in JSON at position 0
    at JSON.parse (<anonymous>)
    at beo-ui.js:69**
(anonymous) @ beo-ui.js:69
postMessage (async)
(anonymous) @ inpage.js:1
u @ inpage.js:1
a.emit @ inpage.js:1
value @ inpage.js:1
value @ inpage.js:1
b @ inpage.js:1
(anonymous) @ inpage.js:1
v.write @ inpage.js:1
y @ inpage.js:1
(anonymous) @ inpage.js:1
s.emit @ inpage.js:1
_ @ inpage.js:1
w @ inpage.js:1
b.push @ inpage.js:1
_write @ inpage.js:1
b @ inpage.js:1
(anonymous) @ inpage.js:1
v.write @ inpage.js:1
y @ inpage.js:1
(anonymous) @ inpage.js:1
s.emit @ inpage.js:1
_ @ inpage.js:1
w @ inpage.js:1
b.push @ inpage.js:1
o._onMessage @ inpage.js:1
postMessage (async)
b._write @ contentscript.js:1
e @ contentscript.js:1
V @ contentscript.js:1
o.uncork @ contentscript.js:1
b._onMessage @ contentscript.js:1
postMessage (async)
o._write @ inpage.js:1
o._onMessage @ inpage.js:1
postMessage (async)
b._write @ contentscript.js:1
b @ contentscript.js:1
(anonymous) @ contentscript.js:1
(anonymous) @ contentscript.js:1
(anonymous) @ contentscript.js:1
1.../../brave/lib/extensionizer @ contentscript.js:1
c @ contentscript.js:1
l @ contentscript.js:1
(anonymous) @ contentscript.js:1
beo-ui.js:166 Preparing menus...
beo-ui.js:332 Menus ready.
beo-ui.js:433 Setting appearance to dark.
beo-comms.js:51 Connecting to hifiberry.local...
hifiberry.local/:1229 **GET http://hifiberry.local/sources/bluetooth/symbols-white/bluetooth.svg 404 (Not Found)**
beo-comms.js:74 Succesfully connected to hifiberry.local.
VM246:140 DomDistiller debug level: 0
beo-comms.js:119 {header: "basicProductInformation", target: "product-information", content: {…}}
beo-comms.js:119 {header: "sources", target: "sources", content: {…}}
beo-comms.js:119 {header: "playerState", target: "now-playing", content: {…}}
beo-comms.js:119 {header: "metadata", target: "now-playing", content: {…}}
now-playing-client.js:354 Loading artwork to view B...
beo-comms.js:119 {header: "setupStatus", target: "setup", content: {…}}header: "setupStatus"target: "setup"content: setupFlow: Array(3)0: {extension: "setup", shown: false, allowAdvancing: true}extension: "setup"shown: falseallowAdvancing: true__proto__: constructor: ƒ Object()__defineGetter__: ƒ __defineGetter__()__defineSetter__: ƒ __defineSetter__()hasOwnProperty: ƒ hasOwnProperty()__lookupGetter__: ƒ __lookupGetter__()__lookupSetter__: ƒ __lookupSetter__()isPrototypeOf: ƒ isPrototypeOf()propertyIsEnumerable: ƒ propertyIsEnumerable()toString: ƒ toString()valueOf: ƒ valueOf()toLocaleString: ƒ toLocaleString()get __proto__: ƒ __proto__()set __proto__: ƒ __proto__()1: {extension: "choose-country", shown: false, allowAdvancing: false, before: Array(3)}extension: "choose-country"shown: falseallowAdvancing: falsebefore: (3) ["network", "sound-preset", "product-information"]__proto__: Object2: {extension: "setup-finish", shown: false, allowAdvancing: true}length: 3__proto__: Array(0)setup: trueselectedExtension: "setup"restartAfter: falsefirstTime: true__proto__: Object__proto__: Object
setup-client.js:64 **Uncaught ReferenceError: historyConstructed is not defined
    at HTMLDocument.<anonymous> (setup-client.js:64)
    at HTMLDocument.dispatch (jquery-1.10.2.min.js:5)
    at HTMLDocument.v.handle (jquery-1.10.2.min.js:5)
    at Object.trigger (jquery-1.10.2.min.js:5)
    at HTMLDocument.<anonymous> (jquery-1.10.2.min.js:5)
    at Function.each (jquery-1.10.2.min.js:4)
    at init.each (jquery-1.10.2.min.js:4)
    at init.trigger (jquery-1.10.2.min.js:5)
    at processReceivedData (beo-comms.js:121)
    at WebSocket.productConnection.onmessage (beo-comms.js:113)**
(anonymous) @ setup-client.js:64
dispatch @ jquery-1.10.2.min.js:5
v.handle @ jquery-1.10.2.min.js:5
trigger @ jquery-1.10.2.min.js:5
(anonymous) @ jquery-1.10.2.min.js:5
each @ jquery-1.10.2.min.js:4
each @ jquery-1.10.2.min.js:4
trigger @ jquery-1.10.2.min.js:5
processReceivedData @ beo-comms.js:121
productConnection.onmessage @ beo-comms.js:113
beo-comms.js:119 {header: "systemVolume", target: "sound", content: {…}}

x838nwy, 2019-12-20:
With specific reference to "setupStatus", once expanded it shows:

{header: "setupStatus", target: "setup", content: {…}}
header: "setupStatus"
target: "setup"
content:
setupFlow: Array(3)
0: {extension: "setup", shown: false, allowAdvancing: true}
1: {extension: "choose-country", shown: false, allowAdvancing: false, before: Array(3)}
2: {extension: "setup-finish", shown: false, allowAdvancing: true}
length: 3
__proto__: Array(0)
setup: true
selectedExtension: "setup"
restartAfter: false
firstTime: true
__proto__: Object
__proto__: Object

Does this help in any way? Let me know if you need anything further.

sodiumlb, 2019-12-20:
I'm also seeing this issue on a clean new sd on pi4 (+ beocreate).

May I suggest HiFiBerry reverts (or add in addition) the download links to images to the previous version while you are investigating this blocking issue.

x838nwy, 2019-12-20:
> I'm also seeing this issue on a clean new sd on pi4 (+ beocreate).
> 
> May I suggest HiFiBerry reverts (or add in addition) the download links to images to the previous version while you are investigating this blocking issue.

Agreed.

BTW on https://www.hifiberry.com/hifiberryos/ it says "Current version: 20191121" but the link downloads 20191219

hifiberry, 2019-12-20:
@sodiumlb @x838nwy The downloads on the web site are now for the November version again until this has been fixed

tuomashamalainen, 2019-12-20:
Thank you all for your reports, they were helpful. I have found the problem and published a fix to the [user interface](https://github.com/bang-olufsen/create). We're checking a couple of other things before reintroducing this update.

Thanks for your patience.

Tuomas from Bang & Olufsen

hifiberry, 2019-12-20:
The new version 20121220 is now online and this bug should be fixed now.

x838nwy, 2019-12-21:
Works on both of my pi4 and Digi+'s just great with one very minor issue - for some odd reason, they're reporting the newest version number, but it still says I need to update. Other than that, everything seems great :D

hifiberry, 2019-12-21:
Hmm, strange. Can you post a picture of the update screen?

x838nwy, 2019-12-21:
> Hmm, strange. Can you post a picture of the update screen?

My apologies. Just tried to replicate it and it didn't happen. Happened for real lat night. May be it's something to do with my browser or something. Terribly sorry.

hifiberry, 2019-12-21:
No problem at all. Maybe a caching issue. I will close this now.

closed 2019-12-21 by hifiberry