hifiberry/hifiberry-os #27: Sound-card DAC+ light no sound settings

dodepier, 2020-01-14:
Hello,

I just installed hifiberry-os on a raspberry pi 3 A+ with a DAC+ light. Works great I have sound. But I am not sure if hifiberry-os did used the correct sound drivers. I have some noise in the output and in the sound config page the sound card is Unknown. There are also no settings available for the sound card. Is this correct?
[image]
This are my diagnostics files:
[amixer.txt](https://github.com/hifiberry/hifiberry-os/files/4060827/amixer.txt)
[aplay.txt](https://github.com/hifiberry/hifiberry-os/files/4060828/aplay.txt)
[arecord.txt](https://github.com/hifiberry/hifiberry-os/files/4060829/arecord.txt)

[df.txt](https://github.com/hifiberry/hifiberry-os/files/4060830/df.txt)

[hw_params.0.txt](https://github.com/hifiberry/hifiberry-os/files/4060831/hw_params.0.txt)

[playerstate.txt](https://github.com/hifiberry/hifiberry-os/files/4060832/playerstate.txt)
[systemctl.txt](https://github.com/hifiberry/hifiberry-os/files/4060833/systemctl.txt)
[top.txt](https://github.com/hifiberry/hifiberry-os/files/4060834/top.txt)
[version.txt](https://github.com/hifiberry/hifiberry-os/files/4060835/version.txt)

hifiberry, 2020-01-14:
That's fine. Sound settings are only available for DSP based cards (Beocreate and DAC+ DSP)

dodepier, 2020-01-16:
Thanks for the quick replay! I have one other question: How can I set a (software)volume limit for the DAC+ light? Since I cannot use the dsptoolkit.

hifiberry, 2020-01-16:
You can't. I recommend just selecting the maximum volume on your amplifier that's connected to the DAC+ Light.

dodepier, 2020-01-16:
Ok I have no volume on my amplifier that is the reason I want a limit. Is the solution to buy a HIFIBERRY DAC+ PRO?

dodepier, 2020-01-16:
> That's fine. Sound settings are only available for DSP based cards (Beocreate and DAC+ DSP)

Maybe it is nice to auto recognise the DAC+ Light when its installed? And change the 'UNKNOWN' to HIFIBERRY DAC+ light. So the user knows that this card has no extra sound options and its installed correct.

hifiberry, 2020-01-16:
Volume limit is only available for DSP-enabled cards. You would need a DAC+ DSP for this.

We'll see that we can improve the text on the sound settings page.

tuomashamalainen, 2020-01-16:
Thanks for reporting this. We'll make this text clearer about what it means in the next update.

Tuomas from Bang & Olufsen

hifiberry, 2020-01-27:
Please try updating to the new version. The DAC+ Light should now be shown as DAC+ Light/Zero/MiniAmp.

kmplngj, 2020-01-28:
`

pcm.!default {
    type plug
    slave.pcm amp
}

pcm.amp {
  type plug
  slave.pcm "plughw"
  ttable {
    0.0=    0.7     #amplication factor
    1.1=    0.7     #amplication factor
  }
}
`
Would something like this alsa mixer config work for limiting the max volume?

hifiberry, 2020-01-28:
You can try, but this will be overwritten by updates as the asound.conf is managed by HiFiBerryOS and not supposed to be changed by the user.

hifiberry, 2020-02-21:
As there was no feedback on this, we'll close this bug. From our point of view, this should be fixed with the January release.

closed 2020-02-21 by hifiberry