hifiberry/hifiberry-os #28: Feature: Master Volume Limit

kmplngj, 2020-01-16:
Hello,

would it be possible set a limit for the master volume (.conf // Web-GUI)? Would be great if the LineOut is connected directly to a power amp.

Thanks
Jan

hifiberry, 2020-01-16:
Master volume can only be limited for DSP sound cards as this requires an independent volume control that can't be changed by the application. This is only available on with DSP enabled sound cards, e.g. the DAC+ DSP.
As each player application controls the volume independently from the GUI, it's not possible to limited this.

closed 2020-01-16 by hifiberry

kmplngj, 2020-01-16:
Would it be possible to set a master volume limit to the sound alsa mixer?

hifiberry, 2020-01-16:
No, the player applications have full acees to the ALSA mixer controls, therefore that's not possible.