hifiberry/hifiberry-os #31: Support for infrared (IR) sensors

santicalcagno, 2020-02-02:
Hi! First of all, thank you for such an amazing piece of software. The experience is really polished!

Currently I have a RPi + BeoCreate board which is connected to a TV via TOSLINK. This TV also has a Nvidia Shield hooked up via HDMI. Unfortunately, the only way to control the volume using the Shield remote seems to be infrared (apart from Bluetooth, but my Pi is a 3B).

- Is there a way to configure LIRC using the standard build of HiFiBerryOS?
- If not, should I create an audiocontrol2 plugin, a custom build of HiFiBerryOS or both? Is there a way to integrate this functionality and make OTA updates not delete it?
- Are you considering adding support for IR in a similar way as the [rotary controller from audiocontrol2](https://github.com/hifiberry/audiocontrol2/blob/master/ac2/plugins/control/rotary.py)? I can also look into making a PR into audiocontrol2 for this if I have the time.

Thanks again for everything!

hifiberry, 2020-02-02:
Hi,

There are no plans from us for this. However, feel free to develop a module for this. You can use the rotary controller module as the base. To test this, you don't even have to do a PR. It's possible to have the module separated and just install it on the system. I will integrate a few changes in AudioControl code and documentation that show you how to load an external module. IT might take some time.
I would recommend to use a pure-python implementation for this as this will be much easier to integrate than adding additional packages to HiFiBerryOS. The official January release does not have 
teh Python GPIO module included. This will come with the February release. To try this, you would have to build your own HiFiBerryOS release based on the 202002 development branch.

Best regards,
Daniel

BTW: If you TV is an LG TV you can try to use LG Sound Sync. It's not yet in the UI, but the backend has the functionality integrated already.

santicalcagno, 2020-02-02:
Awesome news, thanks for the info! Unfortunately it's not an LG TV.
I'll probably wait until the Februrary release is official then, there's a lot to enjoy from this system until then :).
Regards!

hifiberry, 2020-02-03:
The documentation has been been updated now to show how to load a controller plugin dynamically from the configuration file.
https://github.com/hifiberry/audiocontrol2/blob/master/doc/rotary-controller-plugin.md
This should be all information needed to develop you own plugin for audiocontrol.

closed 2020-02-03 by hifiberry