hifiberry/hifiberry-os #34: Airplay not working on RPI 3B

kunaakos, 2020-02-11:
I cannot select airplay as a source on version 20200127 running on a raspberry Pi 3 B (not the plus version). Airplay input can be enabled. I can also connect from my device (mbp running macos 10.15.3), where the output changes to airplay, so everything seems to be working on that end.

hifiberry, 2020-02-11:
I don't understand the problem. You say "output changes to airplay". What exactly is not working?

kunaakos, 2020-02-11:
I mean that I can see the device running hifiberry, and can connect to it using airplay, and my laptop mutes itself as it should when connected:
[image]
... but hifiberryos won't switch over to AirPlay, and I can't seem to switch over manually either, nothing happens if I select airplay from the "start source" modal:
[image]
(spotify is not playing currently)

hifiberry, 2020-02-11:
Try disabling Spotify to check if this changes the behavior.

kunaakos, 2020-02-11:
Disabling spotify by itself did not work, it was still there as an option in the "start source" modal, and the last song played was still there etc. Refreshing etc. did not help, rebooting the device with spotify still disabled did, though! I can now connect and it plays! 🎉
The album art cover is a bit weird though:
[image]
I don't think I have ever listened to this! 🙃

And as a side note, I do NOT have an Apple TV on my home network 😅 It just popped up for a few seconds while I took the screenshot for my previous comment. It shows up and then goes away, have not seen this behaviour before.

hifiberry, 2020-02-11:
Probably something was hanging on the system.

The artwork comes from Audiocontrol that's guessing what song is playing. As airplay reports just "Chrome" as song title, this seems to be the song that's closest. 
If you want to disable this, go to "General/User interface/Download album cover"

kunaakos, 2020-02-11:
Update: I have re-enabled spotify while airplay was playing, to see what happens, and started playing a song. For a brief moment I could hear both airplay and spotify, and after that and spotify stopped playing. The "now playing" modal (not sure how to call it) displayed spotify and the last song played as the current source, but airplay was still playing. When I stopped playing music via airplay, and started again (without disconnecting), I was back to seeing airplay as an input with my friend Trace as a cover.
A few versions back (not sure which unfortunately, but it was before-auto-update) the automatic input switching was working for me.

kunaakos, 2020-02-11:
I think a fix for the album cover issue would be to identify when there's no song info, but yeah, disabling the feature does disable the bug, too.

hifiberry, 2020-02-11:
Can you login via SSH, run "journalctl -f" and switch between Spotify and Airplay and see if there are any error messages (there will be lots of messages).

kunaakos, 2020-02-11:
Okay, so there really is a lot going on in those logs. I grepped the errors, will restart and see the logs again without the cover art fetching noise. What I've seen so far, but this seems like several, separate issues happening at once...

Hardware issues that could be related:

I am getting under-voltage warnings 🧐 I have an APC PowerSurge strip with built in USB ports that should provide 2.4A, and I have the DAC+ pro, nothing else plugged in the pi or the other usb port of the power strip. I do have my amp connected to the same strip, though.

The Pi can't seem to handle an ssh connection and airplay at the same time... There's heavy crackling/jitter and lag. Back to normal immediately after I exit the ssh session.

The rest:

Cover art is still being fetched according to the ui even though I turned that feature off, and cover art is not displayed in the web ui anymore. 

I had to nudge the volume bar in the web UI for inputs to switch over sometimes even if there was nothing playing on the other input. That's around the time I got ALSA errors.

```
Feb 11 10:42:49 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:44:48 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:45:15 HiFiBerry spotifyd[1321]: 10:45:15 [ERROR] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Feb 11 10:45:15 HiFiBerry spotifyd[1321]: 10:45:15 [ERROR] Could not start audio: Alsa error: PCM open failed
Feb 11 10:45:15 HiFiBerry python3[298]: ERROR: mpris - Got empty metadata - what's wrong here? None: None (None) None
Feb 11 10:45:15 HiFiBerry spotifyd[1321]: 10:45:15 [ERROR] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Feb 11 10:45:15 HiFiBerry spotifyd[1321]: 10:45:15 [ERROR] Could not start audio: Alsa error: PCM open failed
Feb 11 10:45:18 HiFiBerry python3[298]: ERROR: mpris - Got empty metadata - what's wrong here? None: None (None) None
Feb 11 10:45:20 HiFiBerry spotifyd[1321]: 10:45:20 [ERROR] Alsa error PCM open ALSA function 'snd_pcm_open' failed with error 'EBUSY: Device or resource busy'
Feb 11 10:45:20 HiFiBerry spotifyd[1321]: 10:45:20 [ERROR] Could not start audio: Alsa error: PCM open failed
Feb 11 10:46:46 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:47:47 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:49:25 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:50:45 HiFiBerry python3[298]: ERROR: alsavolume - webserver@81
Feb 11 10:50:45 HiFiBerry python3[298]: ERROR: alsavolume - <ac2.plugins.volume.http.VolumeHTTPRequest object at 0x74781c50>
Feb 11 10:50:45 HiFiBerry python3[298]: ERROR: mpris - Got empty metadata - what's wrong here? None: None (None) None
Feb 11 10:50:47 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:51:17 HiFiBerry python3[298]: ERROR: alsavolume - webserver@81
Feb 11 10:51:17 HiFiBerry python3[298]: ERROR: alsavolume - <ac2.plugins.volume.http.VolumeHTTPRequest object at 0x74781c50>
Feb 11 10:51:18 HiFiBerry python3[298]: ERROR: alsavolume - webserver@81
Feb 11 10:51:18 HiFiBerry python3[298]: ERROR: alsavolume - <ac2.plugins.volume.http.VolumeHTTPRequest object at 0x74781c50>
Feb 11 10:52:47 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:53:51 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
Feb 11 10:55:14 HiFiBerry sigmatcpserver[243]: ERROR:root:can't get attribute volumeControlRegister from XML
```

hifiberry, 2020-02-11:
Ok, looks like a player isn't stopped correctly. Could be a bug in the pause script.

Try loading this version:

BTW: Most 5V power supplies are NOT suitable for the Pi as it expects 5.1V +/- some percent

kunaakos, 2020-02-11:
Will come back with results of that tomorrow, need to go now unfortunately. Thanks!

hifiberry, 2020-02-21:
As we haven't heard back yet, I would recommend to wait for the next release that should be available next week and test if this works.

hifiberry, 2020-02-24:
New release is out now, can you update and try again?

kunaakos, 2020-02-24:
Hey, sorry for not coming back to this, didn't have time to dive in again.
BUT, I did get a new router and my experience has drastically improved since - my old one was really slow and unreliable. Downloading the new version now!

kunaakos, 2020-02-27:
Everything seems fine now after the update and router change. Thanks tons 💚

hifiberry, 2020-02-27:
Thank you for the feedback.

closed 2020-02-27 by hifiberry