hifiberry/hifiberry-os #35: Builds aren't provided for the Raspberry Pi 2

imgrant, 2020-02-11:
Is it possible to provide releases for the Raspberry Pi 2? Seems like the configs are in place, but without built releases updates won't work.

hifiberry, 2020-02-11:
There are several parts that won't work with the Pi2 as they need CPU features that only Pi3 and newer support. Therefore, we do not build releases for the Pi2 and earlier.

imgrant, 2020-02-11:
Hi, thanks for the info. However, not sure I follow, as you so provide builds for the Pi Zero, which is even less powerful than the 2.

hifiberry, 2020-02-11:
The Zero release only includes Bluetooth and Airplay, no other sources. Doing the same for the Pi2 would make it an Airplay-only system which doesn't make sense in our opinion.
Maintaining multiple releases is a lot of work and we won't spend time on this as there are only few users still using Pi2's. However, you can check out the source and build your own Pi2 release:
https://github.com/hifiberry/hifiberry-os/blob/master/doc/setupdev.md

closed 2020-02-12 by hifiberry