hifiberry/hifiberry-os #36: RPi 3B and AMP+ in boot loop

agentff6600, 2020-02-15:
I etched an SD with HIFIBerry OS and booted it on a RPi 3B with an AMP+.

At the first boot it stopped after 2 min and I couldn't access the RPI via the WIFI AP (didn't tried an ethernet cable yet).

I then disconnected it then from the power and plugged it in again. At first it tried to activate the AMP+ (green light which didn't showed up during the very first boot) with a loud crackle noise from the loudspeakers. 

30 sec in shut down as if there was too much power drawn and tried to boot again. And now it seems it's stuck in that loop. Power up, load a bit, crackle, shut down, power up, load a bit, crackle....

Any idea what might that be?

hifiberry, 2020-02-15:
Is it a 3B or 3B+?

agentff6600, 2020-02-15:
It's a 3B.

hifiberry, 2020-02-15:
Take the SD card out and on your PC create 2 files on the FAT partition named

ssh  
 and
noreboot

Then check if the system boots and you can login via SSH.

agentff6600, 2020-02-16:
thanks! Worked!

hifiberry, 2020-02-16:
Hmm, it's still a bit strange WHY it rebooted. Cat you post the contents of the file /tmp/reboot

hifiberry, 2020-02-20:
@agentff6600 Can you check if you have a file /tmp/reboot and post the content of this file?

pesimon, 2020-02-28:
Hi, I had a similar issue with RPi4 and AMP+ 1.0. 
Initial setup did not work with the MeanWell 60W power supply. 

What I did to get it working: 
- Removed AMP+ hat and booted with official RPi USB-C power supply and wired network
- Configured wifi and rebooted
- added AMP+ and 60W supply again and reboot another time. 

 Now it works :) Thanks for this great project - perfect revival for the AMP+

hifiberry, 2020-02-28:
The Amp2 is NOT compatible with the Pi4. It doesn't provide enough current for the Pi3B+ or Pi4. Powering both the Amp+ and the Pi from separate power supplies can damage the system or one of the power supplies! I strongly recommend not to do this!

pesimon, 2020-02-28:
Sorry for the miss understanding, only had one power supply at a time.
Do you mean Amp2 or AMP+ 1.0 that doesn't have current for RPi4?

hifiberry, 2020-02-28:
I mean the Amp+. The Amp2 is powerful enough for the Pi4. 
Interesting that you got the Amp+ running with the Pi4. There is is good chance this won't run stable.

pesimon, 2020-02-28:
Thx for correction... That might be the reason for inital problems. 
Would be good to add the info about newer models in the [product page](https://www.hifiberry.com/products/ampplus/)

Do you have any specs for the RPi 3 Model A+?

agentff6600, 2020-03-18:
> @agentff6600 Can you check if you have a file /tmp/reboot and post the content of this file?

Sorry for not getting back to you for so long!

The reboot loop happened again after a while and so I could extract the tmp/reboot file. I'll try with the new img (20200226) and see what happens.

agentff6600, 2020-03-18:
works now really well!

hifiberry, 2020-03-18:
Ok, let's close this for now. If you see this happening again, just reopen it and post the contents of /tmp/reboot.
Unfortunately there is no Amp+ test system here anymore.

closed 2020-03-18 by hifiberry