hifiberry/hifiberry-os #37: Pi4 & Digi+

x838nwy, 2020-02-21:
I know using a wired connection is preferred, but just want to let you know what I'm seeing.

Without a wired connection, the boot sequence ends with "No network connected, configure network using wifi setup." Which is usual.

Unfortunately, the unit then does not emit any wifi signal for me to connect to.

This is a clean install and 2 microSD cards, same results. I tried running Rapsbian and wifi works fine.

Thanks

C

hifiberry, 2020-02-21:
Hi,

problem is that we can't debug it without network access as we don't know what's going on. Try connecting it to Ethernet network, then check if it connects to your wireless network.

Best regards,
Daniel

x838nwy, 2020-02-21:
Tried it (wired then connect it to the wifi network). Seems to work fine.
Got an IP address and everything.

hifiberry, 2020-02-21:
So that means you just the access point isn't starting. Unfortunately I don't really know how to debug this at the moment. 
Let's wait for the February release that should be ready next week and see if this changes something (even if there is no known bugfix for this included)

hifiberry, 2020-03-07:
Could you do y test with the February release already?

x838nwy, 2020-03-07:
Hi - super sorry but I'm away from home right now. From what I understand,
it's pretty stable after just a normal update. I'll be able to do this
later next week.

hifiberry, 2020-03-27:
March release is out now. Can you test this one?

closed 2020-04-02 by hifiberry