hifiberry/hifiberry-os #38: MDP not launching from xxx.local

jwrw, 2020-02-23:
Open the interface using http://xxx.local (xxx=<whatever you've called your 'berry) and open up the MDP page from the Sources page.

Click the 'OPEN MDP WEB PAGE' button and it fails to load - on Safari get 'Safari can't open the page xxx.local:9000 because the server unexpectedly dropped the connection.'  Other browsers are similar.

Opening the interface using an IP address e.g. 192.168.1.99 and follow the above steps and MDP page opens just fine.

hifiberry, 2020-02-24:
Yes, it's probably an mDNS thing. Unfortunately these are really hard to debug remotely :(
As you're using Safari, you probably use MacOS - right?

jwrw, 2020-02-24:
Yes, this was on MacOS.  Couldn't get xxx.local working at all from a Windows 10 device - but didn't try very hard.  I did run the 'Discovery' app which shows the services that are being advertised.  There's sftp and ssh (which corresponds to the static set-up I can see on the RasPi) and a couple of other services that must be added in a more dynamic way??  I'll attach a screen grab if that would be useful.

hifiberry, 2020-02-24:
The browser should work just with name resolution, but we can try if advertising HTTP services might improve this.

jwrw, 2020-02-24:
I'd agree - I was a little perplexed myself as by my understanding it should work OK.

hifiberry, 2020-02-24:
Do oyu have another Linux system (even a VM would be fine) around?

Try 
avahi-resolve -n xxx.local

hifiberry, 2020-02-24:
Another idea: does your hostname contain uppercase letters?

jwrw, 2020-02-24:
Tried curl http://xxx.local which worked OK when using the browser didn't.  Did a wireshark and I think it's a IPv6 vs IPv4 thing.  curl uses IPv4 (which works), the browser is trying to do IPv6 and the server drops the connection. So looks like the mdp web server is not doing IPv6 - although mDNS is sending back V4 and V6 responses from the berry.

hifiberry, 2020-02-25:
Good observation. I'll see if there is a chance to have this working at IPv6.

hifiberry, 2020-02-25:
Can you try the following:

- Login via SSH
- Edit /usr/lib/systemd/system/ympd.service
  Change the ExecStart line to
     ExecStart=/usr/bin/ympd -w "[::]:9000"
- systemctl daemon-reload
- systemctl restart ympd

jwrw, 2020-02-25:
Yeah! That's done the trick. Working OK with Safari and Chrome on Mac.  Also "curl -4" and "curl -6" both work OK.  One day I will get my head around IPv6!

hifiberry, 2020-02-25:
Great. We'll integrate this in the next release.

hifiberry, 2020-02-26:
This is integrated into our 20200226 release now. Can you try upgrading?

jwrw, 2020-02-26:
Upgrade all OK.  Launching MPD is fine from the GUI with Safari & Chrome on Mac. Seems OK to close?

hifiberry, 2020-02-27:
Thanks for you help with this.

closed 2020-02-27 by hifiberry