hifiberry/hifiberry-os #44: Airplay dropping

kristojorg, 2020-02-27:
I connect my hifiberry to my network via wifi, and have been experiencing drops in the sound playback. I initially believed this was caused by bad wifi as I was using my ISP provided router. I just ordered a Ubiquiti Amplifi HD router however, and even placed it much closer to the raspberry pi. I'm still getting signal drops however. 

My computer connection to the router seems fine, and so does the hifiberry connection. Both report back strong signals, and the router should be plenty capable of handling audio streaming, especially with only 3 devices connected. 

Can you give me any advice re how to diagnose this?

hifiberry, 2020-02-27:
You might check the airplay logs using

journalctl -u shairport-sync

kristojorg, 2020-02-27:
This is what I'm seeing:
```
Feb 27 09:22:47 soundpi systemd[1]: Starting Shairport-sync...
Feb 27 09:22:50 soundpi systemd[1]: Started Shairport-sync.
Feb 27 09:23:34 soundpi shairport-sync[727]: BusyBox v1.31.0 (2020-02-26 11:50:20 UTC) multi-call binary.
Feb 27 09:23:34 soundpi shairport-sync[727]: Usage: basename FILE [SUFFIX]
Feb 27 09:23:34 soundpi shairport-sync[727]: Strip directory path and .SUFFIX from FILE
Feb 27 09:23:34 soundpi shairport-sync[727]: don't know how to pause
Feb 27 09:23:35 soundpi shairport-sync[727]: BusyBox v1.31.0 (2020-02-26 11:50:20 UTC) multi-call binary.
Feb 27 09:23:35 soundpi shairport-sync[727]: Usage: basename FILE [SUFFIX]
Feb 27 09:23:35 soundpi shairport-sync[727]: Strip directory path and .SUFFIX from FILE
Feb 27 09:23:35 soundpi shairport-sync[727]: don't know how to pause
```
Is there a way to get a longer tail there?

hifiberry, 2020-02-27:
That's all that's there. You might wait until you get drops again and check if you see something.

kristojorg, 2020-02-27:
Ok I just got a drop and logged again, but it's the same result, nothing new : /

hifiberry, 2020-02-27:
Unfortunately that's all that's there. What is the source?

kristojorg, 2020-02-27:
The source is Tidal on a Macbook Pro 15" 2015

hifiberry, 2020-02-27:
I would recommend to connect the Pi the Ethernet and check if you still see the problem.

kristojorg, 2020-02-27:
Alright will do, thank you!

hifiberry, 2020-03-07:
Any update on this?

kristojorg, 2020-03-09:
Ah yes! I upgraded my router and airplay drops seem to be completely gone. Now the only problem I have to solve is occasionally some devices being kicked off spotifyd for some reason. I'll look in to it and open a new issue if I can't figure it out. Thanks for all the work here!

closed 2020-03-09 by kristojorg