hifiberry/hifiberry-os #49: AirPlay stops immediately after start

qerub, 2020-03-08:
As the title says.

Version: 20200226
Worked fine in previous version.

After some troubleshooting I can see that `shairport-sync` launches `/opt/hifiberry/bin/pause-all` (because of `run_this_before_play_begins = "/opt/hifiberry/bin/pause-all shairport"` in `/etc/shairport-sync.conf`) and that `pause-all` actually kills `shairport-sync` (ungrateful child process!).

Enabling `log_verbosity = 1` in `/etc/shairport-sync.conf` and looking in the system journal I see this:

```
shairport-sync[2819]:          0.005948422 "audio_alsa.c:530" alsa: output format chosen is "S32".
shairport-sync[2819]:          0.001842860 "audio_alsa.c:570" alsa: output speed chosen is 44100.
shairport-sync[2819]: don't know how to pause shairport-sync
shairport-sync[2819]:   % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
shairport-sync[2819]:                                  Dload  Upload   Total   Spent    Left  Speed
shairport-sync[2819]:          0.194648187 "dacp.c:157" No DACP port specified yet
python3[509]: INFO: mpris - sent Pause to org.mpris.MediaPlayer2.ShairportSync
shairport-sync[2819]: [158B blob data]
shairport-sync[2819]: ok
beocreate2[573]: Source 'shairport-sync' has activated (index 1).
python3[509]: 127.0.0.1 - - [08/Mar/2020 12:07:36] "post /api/player/pause HTTP/1.1" 200 2
python3[509]: INFO: musicbrainz - found data for Mgła on musicbrainz
python3[509]: INFO: musicbrainz - found data for Age of Excuse V on musicbrainz
python3[509]: INFO: lastfm - Got album cover for Mgła/Age of Excuse from Last.FM: http://coverartarchive.org/release/97171a0b-4f69-492d-849d-afed2533f5df/23991309834.jpg
shairport-sync[2819]: shairport-sync still using the sound card killing it
systemd[1]: shairport-sync.service: Main process exited, code=killed, status=15/TERM
systemd[1]: shairport-sync.service: Succeeded.
```

I find the application logic in `pause-all` a bit hard to follow so I haven't tried to figure out a solution.

qerub, 2020-03-08:
Maybe the solution is simply to set `wait_for_completion = "no"` so that it doesn't pause/stop itself?

qerub, 2020-03-08:
Hmm, seems like you already fixed this in branch 202003: https://github.com/hifiberry/hifiberry-os/commit/2c83eba0fe75593a644f2f809ad62cf8a553313c

Can you explain your branching strategy or point to some notes? Some changes seem to go straight to master and some to these YYYYMM branches. I'm asking because I want to get more involved in the project.

hifiberry, 2020-03-08:
Seems to be a problem with the pause-all script where a process tries to pause itself which should not happend.

Try the following:

edit /etc/shairport-sync.conf and replace 

run_this_before_play_begins = "/opt/hifiberry/bin/pause-all shairport";

by

run_this_before_play_begins = "/opt/hifiberry/bin/pause-all shairport-sync";

Then download this version of the pause-all script:
https://raw.githubusercontent.com/hifiberry/hifiberry-os/202003/buildroot/package/hifiberry-tools/pause-all

and copy it to /opt/hifiberry/bin

hifiberry, 2020-03-08:
Branching strategy is simple:
 master is the latest "official" release, while 20xxxx is the development release.
 In general only critical bugfixes will be applied directly to the master branch (and some
 documentation updates)

 202003 is the release that is planned to be published in March 2020

qerub, 2020-03-08:
Works fine with the changes you provided so I'll close this issue. Thanks!

(Thanks for the branching info too!)

closed 2020-03-08 by qerub

hifiberry, 2020-03-08:
Ok, I will add the change to shairport-sync.conf to the migration of the next release

qerub, 2020-03-08:
Out of curiosity: why does this configuration change have to be handled through a migration and not the normal update process (where all content on the system partition is replaced)?

hifiberry, 2020-03-08:
Configuration files will not be overwritten when updating.  As sahirport-sync is a configuration file, it needs a separate migration (in script/update) to reflext the changes.