hifiberry/hifiberry-os #51: instalation issue (may be)

vsolenoid, 2020-03-10:
Hi,

I have flashed the RPi4 image (on 2 different SDs) and started the Pi. The Pi's LED was blinking for some time, then stopped. I have tried to reach the server by the assigned IP, as I have a couple of other Pi servers and http://HiFiBerry.local wasn't working for me, but "The connection has timed out". I don't have a screen and I don't plan to buy one any time soon. I am using the Volumio image without issues in the same configuration - RPi4 and HiFiBerry DAC+ HW 2.3

How should I proceed if I want to try your OS?

Thanks
Viktor

hifiberry, 2020-03-11:
There might have been a problem with the file system on the latest release. I've compiled a development-release. Can you try this one:
https://www.hifiberry.com/dev/hifiberryos-pi4.zip

vsolenoid, 2020-03-11:
Yeah, it worked. Thanks

closed 2020-03-11 by hifiberry