hifiberry/hifiberry-os #53: Volume jumps

davidchurchill22, 2020-03-18:
Running HiFi BerryOS/BeoCreate/Rpi4/Roon

When I set the volume control in Roon at 50, it jumps to 70.

This does not happen on any other Roon devices

hifiberry, 2020-03-26:
Hmm, I can't see this behavior here. Volume control is working smoothly. Let's see if this still happens with the March release that should be available in a few days.

hifiberry, 2020-03-27:
Please try the latest version from today and check if this still happens

deejgregor, 2020-03-31:
I have a somewhat similar problem where the volume jumps up and usually jumps back down--maybe for a fraction of a second, maybe for a second or two, sometimes it is stuck until I use the volume control. I was able to reproduce this problem on the latest version as of two days ago (I did an online update on my system from  20200226 -- I don't remember if it updated to 20200327 or 20200328). I'm using Spotify and Airplay and I think it's always happened when I'm playing with Airplay from my iPhone. Not using Roon at all.

I didn't want to create a separate issue yet because I'm curious if @davidchurchill22 sees the volume jump up and stay up or if it jumps back down sometimes. I'm happy to make a separate issue if these don't seem related.

hifiberry, 2020-04-27:
@davidchurchill22 Any feedback on this?

davidchurchill22, 2020-05-01:
I have update to the latest version.

Every time i return to this node the volume has moved to max.

i return it to 50%

I come back it is at 100%

hifiberry, 2020-05-01:
Ok, let's try to debug this further as you seem to be the only one with this problem. Exact Roon what other sources are enabled? First disable all other sources then try again.

closed 2020-10-22 by hifiberry