hifiberry/hifiberry-os #55: Need a disable Button for wifi

pos-ei-don, 2020-03-26:
It is urgend to allow to disable the WiFi at all. All those Hifiberry_setup_cdbbb WiFi nets are unsecure and pollute the already filled WiFi frequencies.

hifiberry, 2020-03-26:
The setup wifi will only be started if there is no Ethernet network connected. If your system is running on a stable Ethernet the setup WiFi should not start at all.

pos-ei-don, 2020-03-26:
But i See Them, every Single hifiberry sends out his ssid

hifiberry, 2020-03-26:
Ok, please login via SSH and run the command

ifconfig -a

an post the output here.

tuomashamalainen, 2020-03-26:
You may also check how the system sees the status of ethernet connection. This is used to determine whether or not the setup network should activate.

1. Save the following code as a ".js" file (networktest.js, for example). Place it somewhere in the Raspberry Pi file system.
2. Run `node /path/to/networktest.js`
3. Looking at the output, _connected_ should be _true_. If it is _false_ but your ethernet connection is working, we need to determine why the connection is not detected by this and possibly change how it is detected.

```
process.env.NODE_PATH = "/usr/lib/node_modules/";

require('module').Module._initPaths();

networkCore = require("/opt/beocreate/beocreate_essentials/networking");

networkCore.getEthernetStatus(function(status, error) {
	console.log("Status of Ethernet connection:", status, "Errors:", error);
}); 
```

pos-ei-don, 2020-03-29:
Thank you for your great support.
It seems, that my switsch, a managed switch is booting up too slow.
So after an power-outage, all Wifi-APs are on.
When I manually reboot them, they stay off.
My problem is: when i leave the house, the powersave mode switches off the switch every day...
As I "never!" use the Wifi, I whish to have a setting to disable it completly.
Anyhow, i Checked what happens if the AP is running: It is configured very well. I don't get any rouge DHCP messages into my wired system...

tuomashamalainen, 2020-03-29:
I adjusted the behaviour of the setup access point in the latest update. From now on, the access point can activate within 90 seconds in the following situations:

* the product starts up but there is no network connection
* a Wi-Fi network is removed from the list and the network connection is lost
* the password for a Wi-Fi network is updated and the network connection is lost

Otherwise, it will stay off, even when there is no network connection. 

This is a quick first aid that should help your use case without significantly affecting users who are completely wireless. We will explore adding a switch for completely turning off wireless networking.

Best,
Tuomas from Bang & Olufsen

cvandome, 2020-04-12:
Hello, topic interesting, I understand the behavior for wifi you have implemented, but it is possible to disable both wifi and bluetooth by command line ?

Tradionnally in editing /boot/config.txt and adding this, it should work:
dtoverlay=disable-wifi
dtoverlay=disable-bt

But in this distribution /boot is mounted as read only. Any suggestion please ?

By the way, great work, many thanks !!

pos-ei-don, 2020-04-12:
For me, the current Version works but i'd also favour a Option to comoletely disable it. Maybe it even  an lower the Power comsumption?

hifiberry, 2020-04-12:
You can simply not configure any WiFi network. Also Bluetooth can be easily disabled from the GUI. If you change system config files, you should be aware that this isn't supported and might break things.

hifiberry, 2020-04-27:
Have a look at the latest release 20200424 that now has a "disable WiFi" functionality

pos-ei-don, 2020-04-27:
saw the update for a short periode, but it seems to be dissapeared right now. so update currently not possible.
 will wait for it and try it again. Thank you a lot for your work ans support! Really like it!

hifiberry, 2020-04-27:
Hmm, it's still online. Maybe your system has been updated automatically already. Check the version number.

pos-ei-don, 2020-04-27:
[image]

No, cannot See it. I even rebooted.

hifiberry, 2020-04-27:
That's a bug/missing feature in the March release. Set "Automatic updates" to "Quick", then check again.

cvandome, 2020-04-27:
Great ! it works fine

Many thanks, great job, this music os is actually killing all other os !

hifiberry, 2020-04-27:
Thanks for the feedback.

closed 2020-04-27 by hifiberry

pos-ei-don, 2020-04-27:
Works great for me aswell. Thank you very much.