hifiberry/hifiberry-os #57: No sound since release 20200327.

lost-carrier, 2020-03-27:
Hi

I have a HiFiBerry Digi+ Pro on a Raspberry Pi 3 Model B Rev 1.2 (connected via OPT to some Nubert active speakers if that's relevant) which worked fine under HiFiBerry OS 20200226.

Since the update I cannot hear any music. Using Roon it's hard enough to get the stream starting (sometimes it works, sometimes not). Volume fixed or not makes no difference. Same when I try to play from Roon but via AirPlay.

From iPhone via AirPlay usually starts streaming, but still no sound (Volume full throttle).

Something broken in the new release?

Thanks in advance & regards,
Losty

hifiberry, 2020-03-27:
Might be a bug in the new release. Please send the debug data to support@hifiberry.com

lost-carrier, 2020-03-27:
Hmm... unfortunately collecting diagnostic data does also not work. This spinner just works for ages:
[image]

...anything I could collect via SSH?

hifiberry, 2020-03-27:
I need to check it here first. Easiest thing at this point is to revert back to the previous version. You can either login via SSH and edit /boot/cmdline.txt or edit this file on your PC.

Check the line
root=/dev/mmcblk0p2 ...

if it says "p2", replace it by "p3", if it say "p3", replace it by "p2" and reboot. This will activate the previous version.

lost-carrier, 2020-03-27:
I have a separate SD-card, but nice to know. Thanks!

hifiberry, 2020-03-28:
Please try the 20200328 release that has just been published (you can use the upgrade tool from the UI if your system is running). While the bug in the 20200327 release should not have affected the Digi+, my best guess is that it did. Tested the new release here with the Digi+ and it seems to work. Please give it a try.

lost-carrier, 2020-03-28:
So... 20200327 will not see the update to 20200328, but still report I'm up-to-date with 20200327. Anyhow: I was able to update from an 20200226 directly to 20200328 - unfortunately same behavior again: no sound on Digi+ Pro.

...and BTW I meanwhile found out that even AirPlay from iOS does not actually start streaming (as I wrote in my first post). It's just that the Spotify-App starts counting seconds, but it will stop again after a few seconds passed...

hifiberry, 2020-03-28:
Ok, I don't have a Digi+ Pro here, but tested with a Digi+ without problems. Maybe there is a problem with the detection of the Digi+ Pro. When you scroll down on the main screen, it shows version and the detected sound card. What does is say there?

lost-carrier, 2020-03-28:
Do you mean this?
[image]

BTW: I was now able to create the debug data - E-Mail going out shortly!

hifiberry, 2020-03-29:
It seems to be a bug in the new Raspberry Pi firmware. I'll see if we can either use the old one or find a new one without this problem.

hifiberry, 2020-03-30:
I think, we found the problem. While it's not fully clear how it happens, I think, we have found a workaround that fixes it - at least here in the local test setup.

Try the development release from here:
https://hifiberry.com/dev/hifiberryos-pi3.zip
https://hifiberry.com/dev/hifiberryos-pi4.zip

You need a fresh installation as it changed some things that the updater can't handle today. 

Please let us know if this fixes the problem.

bnilsson11, 2020-03-30:
I tested it on my RP3 Digi+Pro, and now there is light at the end of the (toslink port) tunnel, which was dark before. It plays fine now, thanks.

bnilsson

lost-carrier, 2020-03-31:
I tested yesterday and in-deed: working again! Thank you very much!

Actually I was writing this comment already yesterday... not sure what went wrong or why I don't see it atm (...or maybe: which issue I actually commented on... :-O). Sorry for that...

hifiberry, 2020-03-31:
Thank you for the feedback. We'll publish the new release asap.

hifiberry, 2020-03-31:
Ok, 20200330 release seems to have fixed this. Thank you for reporting it.

closed 2020-03-31 by hifiberry