hifiberry/hifiberry-os #58: Something broken after 20200327

lucasso72, 2020-03-28:
After the update to 20200327 my DAC Pro is not coming up anymore, seems like a problem with network. Strangely, if I connect an hdmi monitor everything works... slower, but it comes up.
Also, as reported by another user, the log collection doesn’t work.

lucasso72, 2020-03-28:
Update: without hdmi connected it does come up but it takes forever (around 10 mins).

hifiberry, 2020-03-28:
There is a problem with this release and we're working on it. Until this has been fixed, we recommend to go back to the previous version

hifiberry, 2020-03-28:
Please try the 20200328 release that has just been published (you can use the upgrade tool from the UI if your system is running). This should fix this bug.

lucasso72, 2020-03-28:
20200328 fixes it. Thanks for the fast resolution!

closed 2020-03-28 by lucasso72