hifiberry/hifiberry-os #61: Unable to turn on Bluetooth on Raspberry 4

kajuku87, 2020-03-31:
When I try to enable bluetooth in the user interface it doenst work.
I'm ussing a fresh image and tried by flashing several times and still not working.
I'm ussing a raspberry pi 4
Thanks!

hifiberry, 2020-04-01:
Looks like a bug. Having a closer looks, this should have never worked. Did you use one of the previous releases already?

kajuku87, 2020-04-01:
No, this is the first release I use. Where can I find the older ones?
Let me know if I can provide you any other useful info.

hifiberry, 2020-04-01:
It was just out of curiosity. No need to test an older one. I'll see that we can package an experimental release today

hifiberry, 2020-04-02:
Ok, there is a new experimental release available for testing.
Please try the following:
- enable SSH
- login via SSH
- /opt/hifiberry/bin/update --experimental

This should fix the Bluetooth audio problem

aatulan, 2020-04-02:
Was experiencing the same problem, updated using the experimental load and the issue has been resolved for me. 
Thanks!

hifiberry, 2020-04-08:
Seems to be fixed in the latest bugfix release.

closed 2020-04-08 by hifiberry