hifiberry/hifiberry-os #64: Allow us to rename bluetooth / sources to allow us to distinguish them from Airplay

steinmb, 2020-04-03:
In MacOS toolbar you can quickly switch output device. Personally if use bluetooth as a fallback for Airplay, especially if I experience audio lag when watching movies. But it is impossible to know which is which from the toolbar.

## Example - Only see two Hifiberry

[image]

At least for the BT device it would be helpful if we could rename it.

hifiberry, 2020-04-03:
You can just edit /etc/bluetooth/main.conf if you want to use a different name for Bluetooth

steinmb, 2020-04-03:
Thank you for extremely fast reply :) Confirm that changing the name there work, also req. a quick reload to make change active with something like 

`systemctl daemon-reload && systemctl restart bluetooth.service` 

But for end users enable users doing this from the UI would be a nice feature to have.

hifiberry, 2020-04-03:
We might think about it, but adding everything to the UI will add complexity and we don't want to end up with an UI as we see it in other distributions where you can change hundreds of settings.

closed 2020-04-03 by hifiberry

steinmb, 2020-04-03:
I totally understand your philosophy. Then perhaps documenting how to change it is a good starting point?

hifiberry, 2020-04-03:
Feel free to add something to the Wiki:
https://github.com/hifiberry/hifiberry-os/wiki

kmplngj, 2020-04-03:
Perhaps just always add a “BT” to the name of the Bluetooth Connection.

steinmb, 2020-04-03:
But if they do that in the build then I think they need to make sure that the broadcast ID is identical to the default name found in `/etc/bluetooth/main.conf` otherwise I think the default behavior can be confusing to regular end users. They would expect that the broadcast id is the same as "mount id" (in lack of a better name)

https://stackoverflow.com/a/54554983 discuss setting an alias playing around with the device name.