hifiberry/hifiberry-os #65: Unable to control volume @ optical input on Rasberry4

chardad, 2020-04-07:
Hi,
I am a user of beocreate4CA and hifiberryOS(20200403). It's a really great music player.  When I use Rasberry3B. Everything is OK.
But when I change to Rasberry4B there are some errors.
- Can't control the main volume @ optical input (I can control volume in volume limit. Other sources are OK.)
- the main volume is less loud then rasberry3.  Rasberry3 is good level @50 but Rasberry4 is good @90
How can I solve these issures?
Thanks in advance!!

hifiberry, 2020-04-07:
Did you re-install the DSP profile after installing the system on the Pi4?

chardad, 2020-04-07:
I do down load HifiberryOS image (hifiberryos-20200403-pi4.img) which was used for rasberry3B.
How could I re-install the DSP profile?

hifiberry, 2020-04-07:
Sound/Advanced/DSP Programs

chardad, 2020-04-07:
I do re-install in DSP Programs / Beocreate Universal / REINSTALL THIS PROGRAM
then DSP installed and restart.
I can control optical source volume in main volume!!  Thanks!!

But less loud problem then RPI3B is not solved.  How could I make louder?

hifiberry, 2020-04-07:
Try rebooting now

chardad, 2020-04-07:
Great!!  I can solve problem after rebooting

Thanks for your great support!

closed 2020-04-07 by chardad