hifiberry/hifiberry-os #67: Connect more than one Spotify account

lnrdnl, 2020-04-19:
In our house we use more than one Spotify account to stream music. I was wondering if it is possible to connect multiple accounts to HifiberryOS. Maybe start a spotifyd service for each account and let the OS determine which one to use?

hifiberry, 2020-04-19:
You don't have to connect ANY account at all. Just don't configure username/password and use it with any account you like.

lnrdnl, 2020-04-19:
Thanks, this is working! But then I don't see the need for the login on the Spotify source page?

When I login with my account, Hifiberry shows up in the Spotify apps of my housemates, but they won't be able to playback music. The app keeps connecting, but nothing will happen.

hifiberry, 2020-04-19:
That's exactly what that does.

closed 2020-04-19 by hifiberry