hifiberry/hifiberry-os #71: 20200330: Update from UI not available.

lost-carrier, 2020-04-28:
I wanted to do a manual update from the WebUI (in order to keep my settings), but "General" -> "Software Update" shows "No updates are currently available".

I'm on 20200330 - according to GitHub releases page there should be 20200403 and 20200424 available.
[image]

hifiberry, 2020-04-28:
This is a bug in the UI. You need to set automatic updates to "quick" or "latest". Otherwise only security updates will be displayed here. This has been fixed in the April version (select "Quick" for this).

lost-carrier, 2020-04-28:
OK... needed a reboot after changing this, but now it works! Thanks!

closed 2020-04-28 by lost-carrier