hifiberry/hifiberry-os #72: Roon “source” disappeared

lucasso72, 2020-04-28:
Hi after upgrade to latest build 20200424 the roon “source”. Reverting to previous version fix the problem, going again to 20200424 and the problem is still there.
Pi3 ++ Hifiberry DAC+

hifiberry, 2020-04-29:
Please login via SSH and post the output of

cat /etc/hifiberry.state

lucasso72, 2020-04-29:
# cat /etc/hifiberry.state
CURRENT_PIMODEL=3A+
CURRENT_MIXER_CONTROL="Digital"
CURRENT_CARD="DAC+ Pro"
CURRENT_HW="card 0"
CURRENT_HW_SHORT=0
CURRENT_SYSNAME="HiFiBerry"
CURRENT_PLAYERS=""
CURRENT_HATINFO="HiFiBerry DAC+"
CURRENT_FEATURES="pi bluetooth pi3orlater"
PIVERSION=3

hifiberry, 2020-04-29:
Ok, I see the problem here. Are you using this with a 3A+ or 3B+?

lucasso72, 2020-04-29:
I'm using the 3A+

hifiberry, 2020-04-29:
Ok, this is a bug in our hardware detection. You can implement a fix by yourself if you like:
Just add arm7 to the FEATURES in line 63 of /opt/hifiberry/bin/reconfigure-players

Then run
/opt/hifiberry/bin/reconfigure-players

lucasso72, 2020-04-29:
I just applied the work-around - it rebooted after running the script - but I still don't see the Roon "source".

 if [[ $modelname == *"3 Model A Plus"* ]]; then
  PIMODEL="3A+"
  PIVERSION=3
  FEATURES="$FEATURES bluetooth pi3orlater arm7"
  return
 fi

I want to add that the first time I upgraded - but I think the problem is still there - every time I move to this version it takes at least two attempts, meaning that after download/apply/reboot the loaded version is still the old one.

hifiberry, 2020-04-29:
That's normal, an upgrade overwrites all files. You first have to upgrade, then edit the file, then run the script.

lucasso72, 2020-04-29:
Nono, sorry, my bad explanation. I did modified and then run the script when I was already with 20200424. I mentioned the problem with the upgrade just as a side-effect in general not in this specific case.

hifiberry, 2020-04-29:
Ok, then delete /etc/hifiberry.state, then run /opt/hifiberry/bin/reconfigure-players and post the output

lucasso72, 2020-04-29:
# ls -ltr /etc/hifiberry.state
-rw-r--r-- 1 root root 255 Apr 29 08:23 /etc/hifiberry.state
#
# rm /etc/hifiberry.state
#
# /opt/hifiberry/bin/reconfigure-players
Sound card settings have been changed, reconfiguring...
Applying settings:
 Players:  (was )
 Raspberry Pi model: 3A+
 System name: HiFiBerry (was )
 Sound card: DAC+ Pro (was )
 HW device: card 0 (was )
 ALSA mixer control: Digital (was )
 Recording:  (was )
 Vendor/product HiFiBerry DAC+ (was )
 Features: pi bluetooth pi3orlater arm7 (was )

Saving settings to /etc/hifiberry.state
Reconfigure features
Reconfiguring Spotifyd
 updated mixer
 updated system name
Reconfiguring Shairport-sync
 updated mixer
 updated system name
Restarting shairport-sync
Reconfiguring Bluetooth
 updated system name
 updated mixer
Restarting bluetooth.service
Reconfiguring squeezelite
 updated system name
Reconfiguring MPD
Restarting mpd
 updated mixer
Reconfiguring DLNA
Reconfiguring Mopidy
sed: /etc/mopidy.conf: No such file or directory
 updated mixer
Reconfigure recording
Reconfigure AudioControl2
 updated mixer
Reconfiguring Beocreate software
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'system'.
 updated card type
Value could not be interpreted as JSON, falling back to old method.
Settings updated for 'sound'.
 updated mixer
["pi","bluetooth","pi3orlater","arm7"]
Settings updated for 'system'.

**reboot**

... and now Roon it's back!

hifiberry, 2020-04-29:
Good to hear. We'll include this fix in the next release.

lucasso72, 2020-04-29:
Thanks for the quick support and fix!

hifiberry, 2020-04-29:
https://github.com/hifiberry/hifiberry-os/commit/d957f45aca32684261b222c8883d82a1e7cf9888

closed 2020-05-28 by hifiberry