hifiberry/hifiberry-os #76: Spotify will not play overnight.

gferrell, 2020-05-01:
I use the HifiBerry RPi4 with the DAC+DSP and HiFiBerry OS. I play music overnight to sleep, but in the morning, it is never playing. I'm not sure how long it plays. I have a Chromecast Audio that I used to use for this purpose and it could play overnight, so I don't believe it is a problem with the Spotify phone app. Perhaps a timeout setting?

hifiberry, 2020-05-01:
There's no timeout. Might be a bug somewhere in the spotify player. You can try logging an and checking the logs with:
journalctl -u spotifyd

Check if you have NOT configured username and password as this is know to crash the player after some time.

xsaranaut, 2020-05-12:
I checked that no user/passwd are configured. The log shows a lot of entries which show a 503 error while fetching metadata (which may be an other issue):

`May 12 09:17:37 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:17:42 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(4)))
May 12 09:17:47 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:17:52 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:17:57 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:18:02 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(14)))
May 12 09:18:08 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(9)))
May 12 09:18:13 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(3)))
May 12 09:18:18 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:18:23 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:18:28 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:18:33 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(13)))
May 12 09:18:38 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(8)))
May 12 09:18:44 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(2)))
May 12 09:18:49 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:18:54 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:18:59 labor spotify-start[503]: Couldn't fetch metadata from spotify: Ok(None)
May 12 09:19:05 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(12)))
May 12 09:19:10 labor spotify-start[503]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(6)))`

When it comes to the restart of the Spotify daemon this is, what my device logs:

`May 12 10:39:21 labor spotify-start[2050]: couldn't parse packet from V4(192.168.172.119:5353): query type 47 is invalid
May 12 10:39:35 labor spotify-start[2050]: Couldn't fetch metadata from spotify: Err(RateLimited(Some(11)))
May 12 10:42:50 labor spotify-start[2050]: couldn't parse packet from V4(192.168.172.119:5353): query type 47 is invalid
May 12 10:43:12 labor systemd[1]: Stopping Spotify connect...
May 12 10:43:12 labor systemd[1]: spotify.service: Main process exited, code=killed, status=15/TERM
May 12 10:43:12 labor systemd[1]: spotify.service: Succeeded.
May 12 10:43:12 labor systemd[1]: Stopped Spotify connect.
May 12 10:43:12 labor systemd[1]: Starting Spotify connect...
May 12 10:43:12 labor spotify-start[3804]: /usr/bin/spotifyd: /lib/libasound.so.2: no version information available (required by /usr/bin/spotifyd)
May 12 10:43:12 labor spotify-start[3804]: /usr/bin/spotifyd: /lib/libasound.so.2: no version information available (required by /usr/bin/spotifyd)
May 12 10:43:12 labor spotify-start[3804]: Loading config from "/etc/spotifyd.conf"
May 12 10:43:12 labor spotify-start[3804]: No proxy specified
May 12 10:43:12 labor spotify-start[3804]: Using alsa volume controller.
May 12 10:43:13 labor systemd[1]: Started Spotify connect.`
Where the .119 IP is the other Hifiberry device.

hifiberry, 2020-05-12:
This shows that Spotifyd was hanging and needed to be restarted. Unfortunately it's not possible to say why it was hanging.

xsaranaut, 2020-05-12:
So there are no other logs enabled, which allow a deeper look into it? Could be that it's the daemon or the underlying librespot?

hifiberry, 2020-05-12:
Could be both. Spotifyd is still known to get stuck and there are several open bug reports that might cause this. Right now, you can only wait...

codingjoe, 2020-05-15:
So I'm getting the same. issue:
```
/usr/bin/spotifyd: /lib/libasound.so.2: no version information available (required by /usr/bin/spotifyd)
```

`spotifyd` crashes, I checked, we are using `libabsound` version 2.0. However, `spotifyd` should be fine with that version.

schwachstelle, 2020-10-19:
[Spotifyd](https://github.com/Spotifyd/spotifyd) doesn't seem to be maintained anymore, so what do you think about switching to [raspotify](https://github.com/dtcooper/raspotify/)? Raspotify doesn't suffer from from this disconnect problem and it even runs on Raspberry Pi 1 and Zero. [Raspberry Pi Audio Receiver](https://github.com/nicokaiser/rpi-audio-receiver) demonstrates that raspotify runs fine beside Bluetooth / AirPlay / UPnP / Snapcast. So for me, raspotify seems like a perfect replacement for spotifyd.

hifiberry, 2020-10-19:
raspotify is simply a script that installs librespot that's the base for spotifyd. spotifyd is still maintained. 
While it seems perfect for you, it's not as librespot doesn't offer an MPRIS interface that's required here.

gferrell, 2020-10-19:
I have found that going to "Airplay" first in Spotify, then selecting HifiBerry fixes the issue.  I have an iPhone so that may only work for Apple devices, but I figured it would help some people.

hifiberry, 2020-10-19:
Yes, this might work, as this doesn't use the Spotify backend, but the music is stream from your mobile device to the HBOS system via Airplay.

hifiberry, 2020-11-07:
Please tyr the new 20201101 release that uses a different Spotify backend now.

hifiberry, 2020-12-10:
Any feedback on the new Spotify backend?

gferrell, 2020-12-10:
Works well for me. Thanks!

hifiberry, 2020-12-10:
Thanks for the feedback. I'll close this now.

closed 2020-12-10 by hifiberry

xsaranaut, 2020-12-10:
In the last 20 hours of Spotify playback almost no problems. No sudden stops, no clicks and pops. So for me it works fine, too.