hifiberry/hifiberry-os #79: spotifyd stops playing every once in a while

xsaranaut, 2020-05-12:
I use the HifiBerry RPi3 with the Digi+Standard and HiFiBerry OS and on another RPi3 with a HifiBerry DAC+ standard. I often play music over the whole day since I'm in home office. Every once in a while it stops playing (same on both devices), the selected speaker in the Spotify app jumps back to "this computer" or "this phone" and pauses. A few moments later the HifiBerry device appears again in the selectable speakers and one can select it again. After that it'll play again like nothing happened (although it sometimes jumps to max volume after this new selection) until every once in a while it stops again. 
Maybe a bug in spotifyd which causes restarts?

hifiberry, 2020-05-12:
Could be a bug in spotifyd. There is already a bug report for this: https://github.com/hifiberry/hifiberry-os/issues/76

closed 2020-05-12 by hifiberry

xsaranaut, 2020-05-12:
Oh, thanks I didn't recognize that issue, my fault. If you need support such as log output or anything I can help with to solve this issue, feel free to contact me.

hifiberry, 2020-05-12:
Checking the log output as shown in the other bug report might help to see what's going on. Just post it in the https://github.com/hifiberry/hifiberry-os/issues/76