hifiberry/hifiberry-os #80: Bluetooth is not working

heikotroetsch, 2020-05-12:
Seems the kernel does not provide /sys/class/bluetooth

`# systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
Condition: start condition failed at Tue 2020-05-12 19:51:47 CEST; 2s ago
           └─ ConditionPathIsDirectory=/sys/class/bluetooth was not met
     Docs: man:bluetoothd(8)`

hifiberry, 2020-05-12:
My guess is that you're using it with a Pi3B which isn't supported.

heikotroetsch, 2020-05-13:
That is exactly right. Why isn't the PI3B supported? It does have a Bluetooth chip. Earlier versions of HifiberryOS did support this function.

hifiberry, 2020-05-13:
The BT hardware on the Pi3 just doesn't allow stable audio connections as it misses hardware handshake. Therefore, we don't support it. You need a Pi 3B+ or Pi 4 for Bluetooth support.

heikotroetsch, 2020-05-13:
Thank you, I think that also cleared up plenty of problems I had in the past with other projects.

closed 2020-05-13 by heikotroetsch