hifiberry/hifiberry-os #86: 20200424 cannot reboot

marcaurele, 2020-05-23:
Hi, following my ticket #85 it looks like it's more an issue with the latest firmware on a Pi4. I can boot it the first time, do the setup, turn on the SSH. Then if I click on restart, the Pi gets again an IP address but neither the SSH server or the HTTP server start again, leaving the system unusable without a reflash of the card.

I went back to the `20200226 and it _restarts_ without any issue.

hifiberry, 2020-05-24:
What do you mean by "latest firmware on a Pi4"?

marcaurele, 2020-05-25:
Sorry not a firmware, but the hifiberry OS `20200424`.

hifiberry, 2020-05-25:
That's strange as this version is running well with lots of other users. However, we'll be releasing a new version probably this wee, you can try this when it's been released.
What sound card are you using?

marcaurele, 2020-05-25:
Order 85334 : with the DAC+ ADC Pro card.

I'll try the new release to see if the problem persists.

hifiberry, 2020-05-28:
I think, we found a bug that might have caused this. Please try the new May release:
https://github.com/hifiberry/hifiberry-os/releases/tag/v20200527

lucasso72, 2020-05-28:
I've the same issue with reboot, also with v20200527.

hifiberry, 2020-05-28:
That's strange. What do you see on the screen after the reboot?

lucasso72, 2020-05-28:
its "unable to connect". Now I tried several times, in some occasions it just takes a lot to come up. I remember it happened already in a previous release...

hifiberry, 2020-05-28:
I mean on a screen directly connected to the Pi

lucasso72, 2020-05-28:
Can't check that right now. Keep you posted later during the say.

hifiberry, 2020-05-28:
ok. You are using an SD card with at leats 4GB - right?

lucasso72, 2020-05-28:
Yep, its 16GB

lucasso72, 2020-06-01:
Sorry for delay.. so I connect the unit to a monitor, it takes quite a bit to boot and now it says "Web interface ready at http://x.x.x.x" but I still get Unable to connect.

hifiberry, 2020-06-02:
You mean you can't connect with your web browser to the given IP address. Can you ping it? 
What Pi are you using?

lucasso72, 2020-06-02:
Yes I can ping it. I’m using a 3A+. 
Anyway now I burned my microSD from scratch with latest build and everything is working fine. Don’t know what to think, something is not ported correctly from one build to another? Why always happen always on Pi3A+? :)
One last thing... I just noticed that with this May build the new feature of displaying the UI on the hdmi screen has been added, but ... on my screen I only see the splash “hifiberryOS” image and a minimal boot sequence that ends with “the UI is available at x.x.x.x”. Should I change anything in any config file?
Thanks.

hifiberry, 2020-06-03:
The local UI via HDMI isn't available on the 3A+ as it just hasn0t enough memory for this.

lucasso72, 2020-06-03:
The pi4b with 2GB is god enough for the UI via hdmi?

usul27, 2020-06-03:
Yes, even the 3B with 1GB is good enough.

lucasso72, 2020-06-03:
Thanks. Anyway personally I cannot reproduce the issue anymore, so unless the fellow who opened this issue has anything to add, from my side this can closed.

marcaurele, 2020-06-03:
I don't have any adapter to connect my pi4 to an external screen. I'll try the new release and give you an update here.

marcaurele, 2020-06-14:
I tried the release using the updater. I struggled to make the squeezelite to work. It did never show up on the squeeze server menu. After any reboot, it didn't keep the settings of the squeezelite.
I have to admit I moved to another distro on my pi4 for now. I have 2 SD cards so I could try later again.
The reboot issue is fixed, but the squeezelite isn't working anymore.

closed 2020-06-14 by marcaurele

hifiberry, 2020-06-15:
You might open a new bug report for the Squeezelite problem. I can only say that there are Squeezelite users also with the latest release. Without further information and debugging I can't say what might be the problem in your setup.