hifiberry/hifiberry-os #92: Add balance support

tinue, 2020-06-01:
In my home office I can't place the loudspeakers equidistant to my desk / chair. It would therefore be great to have support for a "balance" setting, so that I can operate the left speaker at a louder volume. Alsamixer does support this, but I guess I would have to log in via SSH, and I would probably lose the setting after a reboot.
I tried to use "Room Compensation", but apparently it does not measure / compensate for this.

hifiberry, 2020-06-02:
Balance is available for DSP-enabled sound cards, not for other cards. What are you using?

tinue, 2020-06-02:
I am using the Amp2.

hifiberry, 2020-06-02:
There is no integrated balance control on the Amp2. Therefore this would require some ALSA hacking. It's not planned to do this, but if somebody wants to contribute this, we'll look into integrating it.

closed 2020-06-02 by hifiberry