hifiberry/hifiberry-os #94: Make volume slider logarithmic

tinue, 2020-06-01:
About 80% of the volume is barely noticeable, then it's fine for a tiny fraction of the slider, and then it's too loud. I think it would be more intuitive if the slider works in a logarithmic fashion, and thus "feels" linear. I hope I explained this correctly ;-)

hifiberry, 2020-06-02:
It's definitely more complicated than "make it logarithmic". Different applications and sound cards use totally different volume controls and mappings. What sound card do you use and what player do you have problems with?

tinue, 2020-06-02:
I am using the Amp2, and play web radio.

hifiberry, 2020-06-05:
The Amp2 volume control is in fact logarithmic. You simply might not like the full 120dB vole range that it covers. You can limit the range as follows:

Login via SSH, then run

node configure.js sound volumeControlRange [50, 100]

This would limit the slider to the 50-100% range

You need to restart the UI after you did this

systemctl restart beocreate2

tinue, 2020-06-05:
Thanks for the hint. I actually limited the maximum volume using the GUI. Reason is that any playback starts at 100%, and I did not want to risk breaking my loudspeakers. Only after the playback started can I regulate the volume. Maybe a configurable "start volume" would be a good idea?
Would limiting the maximum also impact the way that the volume change is perceived?

hifiberry, 2020-06-05:
HiFiBerryOS itself stored the actual volume and recovers it after reboot. However, if a player changes the volume (e.g. LMS does some ugly stuff), that's something we can't do anything about.
Using the volumeControlRange setting only changes the volume slider in the UI. It basically spreads the range [x,y] to [0,100] in th UI. It does not have an effect on any application that directly controls the volume

closed 2020-06-06 by hifiberry