hifiberry/hifiberry-os #98: Add connectivity to BLE remotes

Laouis, 2020-06-09:
I tried to use an external bluetooth remote with hifiberryOS (like the Beosound essence remote) but they is no pulg and play service :(

Then I found online tutorials suggesting to use 'evdev' Python's module which I can not found in the latest image. 
see here for example: https://raspberry-valley.azurewebsites.net/Map-Bluetooth-Controller-using-Python/

Could we add an interaction settings for the BLE incoming events ? 

If not, could you at least add the module evdev in the hifiberryOS image ?

hifiberry, 2020-06-09:
As there is no Buildroot package for this modules, this will require packaging this first. As this is a very special use case, it will have a very low priority for us. However, if you want to contribute a package, we would include this.

If you want to llok into this, have a look at the documentaion:
https://github.com/hifiberry/hifiberry-os/tree/master/doc

naggie, 2021-03-28:
I've looked into getting `evdev` installed so I can use the rotary encoder running using the `dtoverlay rotary-encoder` mode; I found the GPIO based example unusable as it wasn't able to reliably track the rotary encoder without jumping all over the place.

It's a little involved, but it's certainly possible without rebuilding the entire OS image. I'm writing a blog post about my Hifiberry setup which will include instructions on how to get the device-mode rotary encoder working for a volume control. Until then, here are some commands to build a cpython3.9 compatible wheel for installing via ssh.

Run these commands on another Pi running Raspbian Buster:

    sudo apt-get update
    sudo apt-get install -y build-essential tk-dev libncurses5-dev libncursesw5-dev libreadline6-dev libdb5.3-dev libgdbm-dev libsqlite3-dev libssl-dev libbz2-dev libexpat1-dev liblzma-dev zlib1g-dev libffi-dev

  ```
  wget https://www.python.org/ftp/python/3.9.0/Python-3.9.0.tar.xz
    tar xf Python-3.9.0.tar.xz
    cd Python-3.9.0
    ./configure --prefix=/usr/local/opt/python-3.9.0
    make -j 4
    sudo make altinstall

    # /usr/local/opt/python-3.9.0/bin/python3.9

    /usr/local/opt/python-3.9.0/bin/python3.9 -m pip install --upgrade build

    cd ~
    git clone git@github.com:gvalkov/python-evdev.git
    cd python-evdev
    /usr/local/opt/python-3.9.0/bin/python3.9 -m build

```

The wheel will then be in `dist/evdev-1.4.0-cp39-cp39-linux_armv7l.whl` -- install on the hifiberry with `pip install evdev-1.4.0-cp39-cp39-linux_armv7l.whl`. I don't think that will make it past an OS upgrade though.

closed 2024-01-12 by hifiberry