hifiberry/hifiberry-os #99: Snapserver support for multirrom solutions.

jgabriel98, 2020-06-09:
I think it would be very usefull to add support for snapcast server (not only client support, like it already have).

That way it would be possible to have a multiroom solution, where a main hifiberry is controlling the audio, and the "slaves" just play the song.

is more details:
the hifiberry would have a new feature:
- hifiberry multiroom

When _hifiberry multiroom_ is activated, you would have to set the hifiberrys devices for one of these two modes:
- **main system** or **main player**
- **sub player**

#### The _main system_
The main system would have the _snapcast server service_ running, and also the _snapcast client service_ so it can the audio source too. For the remaining features, they all would behave normally.
It would be the _Node_ to control everything, from audio to UI.

#### The _sub player_
The _sub player_ ones would deactivate all audio source features except for snapcast, and also stop the audiocontrol2 and beocreate UI (web interface) services. In other words, becoming only a snapcast client, not beeing able to control anything (except for local volume).

I've tried to accomplish this feature miself, but i had failed to redirect the alsa audio to a fifo ( so it could be read from the snapserver) only getting a beep sound. Do you guys have any documentation about how the internal flow of audio data occours?

hifiberry, 2020-06-10:
While this is an interesting idea. We probably won't work on this in the near future as HiFiBerryOS was initially designed as a player, not a server. However, we're open to contributions.

jgabriel98, 2020-06-11:
i wouldn't mind to contribute. But i can't understand very well how the internal audio configuration in set.

if you could tell me which files determines the audio behavior i thinks i could do it myself.
in other words, which files (prabably asound.conf files like) configures where the sound comes from and where it goes.

hifiberry, 2020-06-11:
ALSA configuration is in /etc/asound.conf

jgabriel98, 2020-06-11:
is there any other asound.conf, file? for example one in /home/.asound.conf
i've tried to tweak it, but i didn't work. So i tought that maybe other configuration file could overwrite it.

hifiberry, 2020-06-12:
No, that's the file that is used. Note that if you change something, you need to restart the process that's currently playing.

jgabriel98, 2020-07-30:
update: i've given up, since i couldn't redrirect the audio output to snapserver (tried to configure alsa.conf to output to a pipe fifo, but it failed, don't know why since it only failed in hifiberry, maybe there is some hardcoded configs that prevent this manipulation, i don't know)

ssalonen, 2020-08-01:
I am interested in this as well 

@jgabriel98 would you mind sharing the progress here and what did you try to do