hifiberry/hifiberry-os #100: 20200527 instability

hammerfet, 2020-06-10:
The May update seems to be locking up very frequently and requires a physical reboot to recover. 

It’s somewhat random. Everything seems normal for a while then overnight it’s locked up. It also commonly occurs when switching source. In my case between airplay and Bluetooth.

Any ideas how to get logs when this happens? Interface is inaccessible once it goes down. Rolling back always seem to resolve it, but it automatically ends up updating again shortly after (even when set to security). Is there any way to prevent this?

hifiberry, 2020-06-10:
Unfortunately I can't say. Are you using WiFi and BT together? This can result in problem as both are using the same chip.

It should not update automatically if security is selected, this might be a bug.

Please login and post the output of

cat /etc/updater.release

hammerfet, 2020-06-10:
Shows as critical so looks correct. I’m wondering if it’s caching something as sometimes the update is near instant after downgrading. I’ll try a fresh install tonight and see if it makes a difference.

But yes regarding the main issue, I’m using both WiFi and BT on a pi 3A+. Previously this worked flawlessly. Is there a way to externalise one of the radios? Perhaps a dongle based solution?

hifiberry, 2020-06-10:
Ok, this definitely looks like a bug in the updater. Please go back the the previous release an run

cat /etc/updater.time
 /opt/hifiberry/bin/update --check

Until we have a fix for this, you might disable the updater background job using
systemctl disable updater.timer

There's a feature request for an external BT, but we won't have the time soon to work on this:
https://github.com/hifiberry/hifiberry-os/issues/96

hifiberry, 2020-06-10:
You can also add a watchdog to monitor network connectivity by enabling it in 

 /etc/watchdog.conf
[watcher:watchers.internet.Internet]
servers=google.com,facebook.com,alibaba.com
max_fails=3
test_interval=600

This will check if one of these servers is reachable and will reboot if it isn't. While this is a workaround, it should also work for your problem.

hammerfet, 2020-06-11:
Discovered further strange behaviour after some experimenting. Maybe it’s helpful.

As I said my main use case is Bluetooth and Airplay (over WiFi). The lockup will often happens when switching between Bluetooth and airplay. If I restart and only use one, then it’s totally fine until I disconnect or turn off the device that is connected.

It’s especially noticeable on Bluetooth that after every disconnect/reconnect of Bluetooth from the same device, the latency gets longer and longer. Going from around 300ms after reboot to over 1s after a bit of switching. Eventually it will lock up.

Airplay on the other hand becomes more and more unstable and will disconnect more frequently. When it does, the web interface also locks up. Sometimes it can recover after a few mins, sometimes not and a reboot is needed.

I’ve disabled the updated to 20200527 for now and all is back to normal. I did notice this behaviour somewhat on 20200424, but it’s no way near as sensitive. Takes a lot of stressing to upset it. Most of the time is have around 320ms Bluetooth latency. This is my primary use case, hence I’m keeping such a close eye on it

hifiberry, 2020-06-11:
You will probably need to stay with the 202004 release for a longer time as we can't control or debug the internals of the Bluetooth stack. We try to keep the software updated with newer releases. There might be some version in the future that might improve this, but that's all I can say about it.

mkulichkin, 2020-06-21:
I confirm this happens to me also.

I'm on Raspberry pi 3b+ using DAC + DSP and wi-fi connection.

Web interface and UPnP become inaccessible although the interface works on the display and mouse connected to PI.

Music still plays when that happens: In the case of UPnP till the end of the track and for internet radio it continues.

I usually reboot the system, but the last time it came back itself after some time.

hifiberry, 2020-06-22:
Try disabling WLAN power management by logging in and running

 iwconfig wlan0 power off

hifiberry, 2020-06-30:
Another workaround is configuring the watchdog. Just edit /etc/watchdog.conf like this:

[watcher:watchers.internet.Internet]
servers=google.com,facebook.com,alibaba.com
max_fails=3
test_interval=300

This will test network connectivity every 5 minutes.  It might help if your AP drops connections of systems that are not sending any traffic.

hifiberry, 2020-07-21:
The next release will allow to use an USB BT stick for Bluetooth connectivity.

closed 2020-10-13 by hifiberry