hifiberry/hifiberry-os #101: BLUETOOTH ON HIFIBERRYOS

armatron, 2020-06-10:
Problem: 
It happens that windows automatically connects to BT. If I want to stream music from my phone, I need to turn off BT on my wife's laptop first.

Feature request: 
to be able to somehow disconnect connected BT device from HifiBerry OS. My proposal is to add disconnect option in Sources \ Bluetooth near enabling BT or at least interact action (triggered by url command).

Thanks for your effort.

hifiberry, 2020-06-10:
I'm not sure if a "disconnect" option would help as your Windows system might just reconnect later.  You might try the following: disable BT, then re-enable it again. Does your Windows system stays disconnected?

hifiberry, 2020-07-14:
Did you do the check above?

hifiberry, 2020-11-07:
Closing due to inactivity

closed 2020-11-07 by hifiberry