hifiberry/hifiberry-os #104: Bluetooth on 20200527

armatron, 2020-06-12:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
When RPI is connected to router via UTP cable, WiFi connection is disabled and my phone connects to HifiBerry OS via BT, volume on HifiBerry does not change. When I start to play in my client, volume in HifiBerry OS changes to the same level as client (50%).

This seems OK to me. 

But:

when RPI is connected to router via WiFi and client connects to HifiBerry OS, volume on HifiBerry changes to 100% but volume on my client remains 50%. Also change volume on my phone don't affect volume on HifiBerry OS and vice versa. BT starts to periodically disconnecting and reconnecting. 

I noticed this issue in Hifiberry OS version 20200527. 
HifiBerry OS version 20200424 with the same scenario (with WiFi or with UTP cable) and same HW and client  has no issues. So I assume, this is not HW issue but issue in HifiBerry OS version 20200527.

**HiFiBerryOS version**
20200527

**HiFiBerry sound card**
Amp2

**To Reproduce**
Steps to reproduce the behavior:
1. Set wifi connection in hifibery os. 
2. Connect rpi to router via utp cable.
3. Disable wireles in hifibery os. This setup wil enable to set BT volume, which will be tested in the following steps. 
4. Pair and connect client via BT to hifiberry os. 
5. Start to stream music on client.. This will probably change volume in hifiberry os. 
6. Set volume in hifiberry os to 50%.. The volume on cliet should change automaticaly to 50%. Now, we got synchronised volume. This is expected behaviour. 
6. Stop streaming and disconnect BT connection. 
7. Remove utp cable from rpi and wait, till hifiberry connect to wifi. This setup should produce issue in the following steps. 
 8. Connect the same client to hifiberry osvia BT. 
9. When clien is connected, volume slider in hifiberry os automaticaly moves to 100%. If not, it will move to 100% when start streaming. 
10. Change volume slider in hifiberry os. Volume does  not change on client. 
11. Change volume on client. Volume on hifiberry os does not change.
12. When not streaming, BT connection periodically reconnects. 

**Expected behavior**
1. BT connection shoul be stable also if not streaming. 
2. Volume slider should be synchronised also when rpi is connected to wifi and utp cable is disconnected. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 Different os and browsera has  no impact. 

**Additional context**

hifiberry, 2020-06-12:
This is probably something that we can't fix by ourself as it looks like a low-level problem with WiFi/BT interaction. If somebody has more knowledge about this, any help is welcome.

armatron, 2020-06-15:
I test the same scenario with the same rpi and client but with different hifiberry os version 20200424. There are no BT issues. Maybe this helps.

hifiberry, 2020-06-15:
Unfortunately it doesn't ;-) 
The whole buildroot set have been upgraded from April to May release. Therefore it's almost impossible to say what specifically is causing it :(

hifiberry, 2020-06-22:
Not sure if this helps, but try disabling WLAN power management:
 iwconfig wlan0 power off

armatron, 2020-07-14:
_iwconfig wlan0 power off_
did not help in version hifiberryos-20200701-pi3 neither in version hifiberryos-20200708-pi3. 

As soon I connect to wifi in HiFiBerryOS, I faced with issue mentioned above.
If I remove connected connection from wifi in HiFiBerryOS  and restart HiFiBerryOS, then it works without issues.

hifiberry, 2020-10-22:
Did you try one of the releases from the last months that now supports external BT sticks?

armatron, 2020-10-22:
Hi,

My current setup is Raspberry 3b+ +, Bluetooth dongle, wifi enabled.
I still have a "problem" with volume but this is a "problem" with my phone. Once my phone is connected to BT, BT media volume goes to 50% which put HiFiBerry volume slider to 90% when I start playing music.
But when changing volume slider on HiFiBerry or phone, volume is synchronised, which seems ok.

BDW: volume slider on my phone is linear, but volume slider on HiFiBerry seems exponential (50% on my phone is 90% on HiFiBerry, also when I change slider on HiFiBerry in lower volume it has barely effect to volume but if I change HiFiBerry slider on high volume, it effects massively to volume). 

Is there any change to set volume linear?
Thanks.

hifiberry, 2020-10-22:
No, there is no way to change the volume control. There isn't just "logarithmic" and "linear", but all kinds of curves. As there are no standards for this, there is no way to find a configuration that will work find in with all players and sources. It would be nice if anybody would stop using percent values for volumes and instead use decibels. However, I don't think this will ever happen.

I'll close this bug report now.

closed 2020-10-22 by hifiberry

armatron, 2020-10-23:
Hi, I have made a few more tests. I have installed a BT widget on my phone which enables to set BT media volume after connect. In this widget, I. set volume to 0%.

Once my phone is connected to HiFiBerryOS, volume slider goes to 0% (set by widget) but right after that, volume on my phone goes to 50%, no matter of HiFiBerryOs volume slider position. It seems that HiFiBerryOS set volume on my phone after BT connection is established. 
I noticed the same behaviour on my RPI 3B+ with internal BT receiver and WIFI disconnected (network cable)

I made the same test with my phone and some BT portable speaker, but in this case, BT media volume on my phone remains 0%.

Is volume on my phone set by HiFiBerryOS after BT connection established?

hifiberry, 2020-10-23:
Yes, HiFiBerryOS has to synchronize the local volume with the BT sender. If you change the volume on HiFiBerryOS it has to send this change back to the BT sender. That's how it needs to be. Otherwise you would end up with two independent volume controls.

armatron, 2020-10-23:
Hi,
I understand that but the problem is, that HiFiBerryOS set volume slider to 50% each time, I connect my phone via BT, no matter of current HiFiBerryOS volume slider position. And this makes volume in my speaker too loud with AMP2.

glerak, 2025-03-23:
I prefer to adjust the volume only from the hifiberryos UI (including web, API, serial link, etc.). None unpredictable volume adjustments from a BT device. Could this be done? Any setting option?
Could it be a solution?

(https://support.hifiberry.com/hc/en-us/community/posts/5442991714961-Bluetooth-volume-issues?page=1#community_comment_26062773815581).