hifiberry/hifiberry-os #105: Spotify Connect: "spotify is not active on this device"

iwyg, 2020-06-12:
I ran into this issue recently and I'm a bit puzzled why this is issue suddenly occurring. The hifiberry-os device is showing in the device list on every spotify client, but it is disabled saying "spotify is not active on this device".  It's weired because it just worked fine couple of days ago. 

The problem goes a way once I've configured a spotify account in the `/etc/spotify.conf` though this shouldn't be necessary at all. 

I can reproduce this behaviour for versions 20200424 and 20200527.
This is probably not directly related to hifiberry-os but to spotifyd, but since my other device, which runs volumio + spotify connect plugin just works fine, I'm kinda out of ideas.

hifiberry, 2020-06-12:
You probably should address this directly on the spotifyd repo:
https://github.com/Spotifyd/spotifyd

Note that configuring username/password isn't recommended as the metadata interface of spotifyd crashes regularly in this case (there is an open bug about this) and need to be restarted.

iwyg, 2020-06-17:
Will do. For now, this issue fixed itself through magic :|

closed 2020-06-17 by iwyg