hifiberry/hifiberry-os #106: WiFi PowerSave disconnects RaspberryPi

PatrickDrewsES, 2020-06-16:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
WiFi is disconnected on a regular basis. I assume its the power-saving feature of the pi 4. However, PowerSave should be turned off by default.
/etc/network/interfaces.bak do not include a command for turning power save off.

**HiFiBerryOS version**
20200527

**HiFiBerry sound card**
DAC+ pro

**To Reproduce**
do nothing (idle leads to power save)

**Expected behavior**
should be alway on line

**Screenshots**
none

**Browser (if applicable)**
not applicable

**Additional context**
infrastructure is a fritz!box network with mesh repeater

PatrickDrewsES, 2020-06-16:
was easy since iwconfig is included:

`iwconfig wlan0 power off`

sorry for bothering you, guys.

closed 2020-06-16 by PatrickDrewsES

fredrike, 2022-05-18:
@PatrickDrewsES how did you put `iwconfig wlan0 power off` to run after a reboot?