hifiberry/hifiberry-os #107: DSP volume not available

NearNetto, 2020-06-17:
**Describe the bug**
DSP volume unavailable from Roon - only fixed and device volume is possible. Only DSP volume provide 24 bit output with dither as recommended by Roon.

**HiFiBerryOS version**
Software 20200527

**HiFiBerry sound card**
Digi+ at RaspBerry Pi model 4b.

**To Reproduce**
Roon settings: Select volume mode.

**Expected behavior**
Choice between DSP and fixed volume.

**Browser (if applicable)**
 Same issue with Roon at iPad and Win10.

**Additional context**
DSP volume is or has been available at Raspberry Pi model 3 with the same Digi+ HAT running HiFiBerryOS.

hifiberry, 2020-06-17:
Are you sure you're running the same HiFiBerryOS version on the Pi3?

NearNetto, 2020-06-17:
No, I only have Model 4b available, but I have got screenshots and a detailed description from Benchmark Media in US last night, where they confirm they are running Model 3b with Digi+ HAT. Here the DSP volume feature works as predicted without any issues. They might run an older version of HiFiBerryOS though. In fact I think they are...
Michael

hifiberry, 2020-06-17:
Ok, that might be some bug that had been introduced in one of the latest versions. If you're only using Roon, you might try an older version as a workaround:
https://github.com/hifiberry/hifiberry-os/releases

NearNetto, 2020-06-17:
I just tried the 20200226 OS and it works as a workaround – the DSP volume is available here!

So – it is a bug introduced in a later versions this year!

Best regards

Michael

 

https://github.com/hifiberry/hifiberry-os/releases

NearNetto, 2020-06-17:
Please notice that the volume is labelled “Device” via Roon even when it is in DSP mode. You can tell the difference in several ways, but the most obvious is that in the correct mode (DSP 24 bit with dither) the volume steps are calibrated in dB.
Michael

hifiberry, 2020-07-14:
We had some technical discussions with Roon about this. In general, DSP volume control is simply incompatible with the multi-service approach were using in HiFiBerryOS. Enabling this will break volume control for other services which isn't what we want to do. This might be ok for a Roon-only setup, but as HiFiBerryOS isn't a Roon-only distribution, we won't integrate this officially. There might be a way to hack the configuration (we need to look if this is possible), but there won't be any official support from us for this.

hifiberry, 2020-07-15:
In the next release (end of July), you can create a file /etc/hifiberry_raat.softvol that will enable softvol in RAAT. This isn't officially supported by us and we don't recommend to use it as it is incompatible with volume control of other players. However, feel free to try it out.

hifiberry, 2020-07-15:
https://github.com/hifiberry/hifiberry-os/blob/master/doc/hacks.md

hifiberry, 2020-07-15:
https://github.com/hifiberry/hifiberry-os/commit/26c71a08a6858c2e303255e1f403d12f7a0bcf26

hifiberry, 2020-07-15:
This will be included in the next release.

closed 2020-07-15 by hifiberry