hifiberry/hifiberry-os #109: wifi has turned itself on and created the setup access point

Farcus2000, 2020-06-22:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

wifi has turned itself on and created the setup access point

**HiFiBerryOS version**
You find this on the "About" screen. It looks like Software 20YYMMDD
20200527

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".

DAC+ Pro

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
Something happenes about every 7 days

**Expected behavior**
A clear and concise description of what you expected to happen.

With wifi disabled and setup complete I do not expect the setup wifi basestation to spring back into life.
[hifiberry-debug-2.zip](https://github.com/hifiberry/hifiberry-os/files/4813065/hifiberry-debug-2.zip)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Logs have many recurring issues which may be responsible for causing this.

hifiberry, 2020-06-22:
Thanks for reporting this. WiFi AP should only be enabled if no network is detected after a reboot. Is it possible that there was a power outage when this happened?

Farcus2000, 2020-06-23:
Agree a possible cause. No outage today or last week with same symptom.

hifiberry, 2020-06-30:
Unfortunately I can't say why this happend. Try the latest release and make sure WiFi is deactived in the network settings.

Farcus2000, 2020-07-01:
Update installed overnight without a hitch. The access point activated again Saturday, about a week apart. 

All the configuration settings retained and confirm wifi disabled. Thank you.

hifiberry, 2020-07-14:
Did this happen again with the latest release?

Farcus2000, 2020-07-15:
Thank you for checking back. 

Afraid so. Most recent was overnight 15th July no power or internet outage.

Last week 8th was not clear as there was an interim OS update as well as a maintenance outage on my ISP and the WAP activated. I didn't flag it as the ISP being down was a new dimension.

Loving the sound, particularly notice voice. Roon trial underway and works really well.

Thanks again.

Mal

hifiberry, 2020-07-21:
Ok, in the next release, we'll add a way to fully disable this. You can create a file "noap" on the FAT partition of the SD card (either on your PC or on the system in /boot). If this file exists, the access point will never be started.

https://github.com/hifiberry/hifiberry-os/commit/97c4e801710209e9acba07f07ef9b813342e05bc

MattHardwick34, 2020-08-20:
Trying to create noap file results in a message about read only file system. Why isn't there an option to disable this feature though the configuration?

hifiberry, 2020-08-20:
You need to remount the file system read-write

mount -o remount,rw /boot

We won't add UI configuration options for every feature - we don't want to end up like other distributions with hundreds of settings in the UI.

MattHardwick34, 2020-08-20:
OK that worked thanks - we'll see how things go after a week or two!
And that makes sense but shame we can't get to the bottom of why it's randomly starting!

closed 2020-11-01 by hifiberry

EpicLPer, 2023-08-03:
> Ok, in the next release, we'll add a way to fully disable this. You can create a file "noap" on the FAT partition of the SD card (either on your PC or on the system in /boot). If this file exists, the access point will never be started.
> 
> [97c4e80](https://github.com/hifiberry/hifiberry-os/commit/97c4e801710209e9acba07f07ef9b813342e05bc)

I just want to throw in my few cents and say that this is a really bad decision... for over a year now I had various network problems and never could quite figure out why, until I ran a Wireshark trace today and finally figured out that HiFiBerryOS throws in random DHCP offers...
I've had servers on my network go down because it assigned IPs that were not available in the DHCP range, had mobile phones ask for guest logins that didn't exist due to this and it caused quite a few awkward moments at home due to this...

It'd be great to at least mention in the setup guide that this could be a problem and that this option exists, so others with less knowledge don't suddenly have to debug this on their own.

tompeel, 2023-10-23:
I agree, this has taken me a long time to get to the root of quite a few problems. Its an incredibly dificult thing to track down especially for people who dont know how to use a complicated tool like Wireshark (I do) I spent over 18 months thinking is was an issue with a Unifi certificate as all the phones would get DHCP from this.
Can it not jsut be hard disabled unless you enable AP mode... i never used AP mode and didnt even know it was a thing?! No UI option for AP mode ether i dont think but somewhere this needs shutting down.