hifiberry/hifiberry-os #112: New 96Khz profile sound tinny and lifeless

MrRobotoGit, 2020-06-29:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
Loaded the new DAC+ DSP profile after upgrade and something is wrong. Sound is thin and tinny, midrange is poor. I have my own 96/192Khz profiles made from Beocreate Universal 10 and it does sound much better and fully.

**HiFiBerryOS version**
Software 20200629

**HiFiBerry sound card**
DAC+ DSP

**To Reproduce**
Just load the program 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Browser (if applicable)**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

hifiberry, 2020-06-30:
Just measured the profile here and frequency response is flat.  Do you have any filters configured?

[image]

hifiberry, 2020-06-30:
Do you use the analog or digital output?

MrRobotoGit, 2020-06-30:
I use analog output to a tube preamp. The Hifiberry is calibrated using REW. My DSP programs (96/192Khz) have been made by modifying the original V10 beocreate 48Khz profile. After that all the profiles had the same frequency response. So, why a new DSP profile must change the frequency response? I don't really get it. How did you take those measurement?

Also, why not include a 192Khz profile by default which is just a matter of changing a couple of things in Sigma? I own the DAC+ DSP since March, and I like the HiFiBerryOS,  but I keep finding bugs and bugs between the monthly versions. Do you really test things before release them? Do you think I am ok to recalibrate my audio system at any DSP program release??

hifiberry, 2020-06-30:
Feel free to use the old DSP profile if this works better for you:
https://raw.githubusercontent.com/hifiberry/hifiberry-os/master/buildroot/package/dspprofiles/dspdac-10.xml

You don't have to upgrade the DSP profile if you don't want this.

You can also set "automatic updates" to security and you won't get any automatic updates - except security updates - if this works better for you.

MrRobotoGit, 2020-06-30:
I don't really know if I want the new profile since you don't post anywhere what's changed between V10 and V11. It was funny to spend time on this during the lockdown, but now I just want to come back from work and put some music on. And this "don't use it if you don't want it" doesn't make sense. Win2000 was fine so why update. Bah.

closed 2020-06-30 by hifiberry

MrRobotoGit, 2020-06-30:
Oh and it seems you don't even read.

hifiberry, 2020-06-30:
We'll add the old v10 profile to the next release and also clarify the difference betwen 48kHz and96kHz profiles