hifiberry/hifiberry-os #113: Screensaver

armatron, 2020-07-02:
**Describe the solution you'd like**
It would be great feature to standby monitor/touch screen after 5 min after music is stopped;  power up monitor/touch screen on touch or activity from web interface

**Describe alternatives you've considered**
One alternative is to set screen black (screensaver) but better solution would be to power screen off (standbye), only touch function stays active, if this is possible at all. 
Also, providing option to enable/disable screensaver would be great.

hifiberry, 2020-07-02:
Thanks. This is on our idea list already.

skumka, 2021-01-14:
I would love to power screen off (standby), only touch function stays active.
If it comes to screensaver it would be nice to have a 'clock' option.

birkb, 2021-01-30:
I have sucessfully tested this manual screensaver workaround for a Raspberry 7" display:
- `mount -o remount, rw /boot`
- add `consoleblank=60` to /boot/cmdline.txt
- reboot
- General > Display > Use Connected Display = disable
[image]
The blinking cursor will appear and after 60 seconds the display will be blanked. The local ui will come back after you have switched to `Use Connected Display = enable` in the web ui. Not nice, but at least a start...

tysonnorris, 2021-02-24:
I have been prepping a PR to https://github.com/bang-olufsen/create where the ui-settings extension has a fake screensaver behavior - is this something that would suffice? 

For background, thinking about this In general, it seems strange to completely disable the display:
* if the display is the only control, will tapping it clear the screen saver e.g. if the screen is disabled completely?
* if the display is not the only control (I use my phone as well), I want the connected display to update (break out of screensaver) if e.g. transport controls change play/pause, or now-playing changes, or basically any event, without having to touch it. 

In my setup, there is just a html overlay with a simple css animation loop that comes into view after a timeout. 
It works like:
* will only show on localbrowser (connected display), not other http clients
* most (all?) beo events will cause it to hide and reset the timer
* there's an adjustment slider to set the timeout between 1min and 10min

My goal is not really to save power, but just prevent screen burn if the system is on but idle. To save power, I would suggest more like an "auto-off" that completely turns off the system, especially if the system is not usable once the screen is disabled. One variant I've also thought about is dimming the screen - I know some dsi displays have this feature (not sure about the raspberry pi display), which could be controlled in coordinated fashion to the browser overlay activation/deactivation (but trigger an os script to dim the backlight). 

When I get the beo-create PR ready I'll add a link here.

schmurtzm, 2022-01-18:
[The screensaver from Material skin](https://github.com/CDrummond/lms-material/blob/437c5b3cde96b84caaedd4079238af6485a175a8/MaterialSkin/HTML/material/html/js/screensaver.js) (a skin for squeezebox servers / Players) is really great :  it display a clock with date, moving mouse ou touch over it to quit. It could be a great source of inspiration.

hifiberry, 2022-08-04:
This has been implemented some time ago (even thought there's a bug in the current release: https://github.com/hifiberry/hifiberry-os/issues/357)

closed 2022-08-04 by hifiberry