hifiberry/hifiberry-os #114: Spotify playback not showing in web interface

danvincent, 2020-07-02:
DO NOT remove the blocks below, but fill these with the requested data. Incomplete bug reports will be ignored!
You should remove this comment before posting the bug report.

**Describe the bug**
A clear and concise description of what the bug is.

**HiFiBerryOS version**
20200701 and 20200621

**HiFiBerry sound card**
Post the exact type, e.g. DAC+ Pro instead of "DAC+".
DAC+

**To Reproduce**
Play a track on Spotify and transfer playback to the HiFiBerry

**Expected behavior**
Sources should display track details and artwork (where available)

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Browser (if applicable)**
 - OS: Android 10
 - Browser: Chrome

**Additional context**
Appears to have started in 20200621

hifiberry, 2020-07-02:
Thanks for reporting this. This is really a new bug that has been introduced with the work on the local music library. Bugfix will be included in the next release.

hifiberry, 2020-07-02:
https://github.com/hifiberry/audiocontrol2/commit/0cbe8f3c273f023227f4ee8af92862891a90165a

hifiberry, 2020-07-10:
Can you check the latest version? This should be fixed already.

danvincent, 2020-07-10:
Yep fixed in the last version ✅

danvincent, 2020-07-10:
✅✅✅

closed 2020-07-10 by danvincent